openwrt/packages #2407: cups is missing doc and ipptool folders

futurechild, 2016-02-22:
cups is missing the following folders:
-doc containing cups.css, index.html etc.
-ipptool containing a few testfiles.

The below changes to the Makefile solve this.

changing this:

```
    $(INSTALL_DIR) $(1)/opt/share/cups
    $(CP) $(PKG_INSTALL_DIR)/opt/share/cups/templates \
        $(1)/opt/share/cups/
```

into this:
```

    $(INSTALL_DIR) $(1)/opt/share/cups
    $(CP) $(PKG_INSTALL_DIR)/opt/share/cups/templates \
        $(1)/opt/share/cups/
    $(CP) $(PKG_INSTALL_DIR)/opt/share/cups/ipptool \
        $(1)/opt/share/cups/
    $(INSTALL_DIR) $(1)/opt/share/doc
    $(CP) $(PKG_INSTALL_DIR)/opt/share/doc/cups \
        $(1)/opt/share/doc/
```

Have these folders been ommited intentionally ?

hnyman, 2016-02-22:
I think that cups is still in located the "oldpackages" repository and currently unmaintained.
http://git.openwrt.org/?p=packages.git;a=tree;f=net/cups;hb=HEAD

Feel free to assume maintainership for the package and fix it, and then create a pull request to import it here.

closed 2016-02-22 by hnyman

psycho-nico, 2016-02-22:
Just to let you know that I'm still working on reviving cups...

luizluca, 2016-02-23:
@psycho-nico ,  just to let you know, I have the hplip package ready just waiting the cups package.  I'm interested only in hplip-sane but libcups is a hard build requirement.

futurechild, 2016-02-23:
@psycho-nico , Did you find the cups version 2.1.2 Makefile currently used by Entware-ng ?
They pointed me here for changes as they just sync their repo with OpenWrt but I can only find a Makefile for version 1.5.3.

luizluca, 2016-11-18:
@psycho-nico, is there any cups update?

anarcat, 2018-09-13:
i have opened an issue about missing cups support in https://github.com/openwrt/packages/issues/7016

In the meantime it seems the canonical feed is https://github.com/Gr4ffy/lede-cups/ but no automated build is currently provided.