openwrt/packages #2408: Problem adding package uwsgi

simmyg89, 2016-02-22:
Hi all,

i'm trying to add the uwsgi package but this won't compile
this is the Makefile

``` 
include $(TOPDIR)/rules.mk
 
PKG_NAME:=python-uwsgi
PKG_VERSION:=2.0.12
PKG_RELEASE:=1
 
PKG_SOURCE:=uwsgi-$(PKG_VERSION).tar.gz
PKG_SOURCE_URL:=https://pypi.python.org/packages/source/u/uWSGI/
PKG_MD5SUM:=1451cab954bad0d7d7429e4d2c84b5df
 
PKG_BUILD_DIR:=$(BUILD_DIR)/uwsgi-$(PKG_VERSION)
PKG_BUILD_DEPENDS:=python libopenssl libpcre
 
include $(INCLUDE_DIR)/package.mk
$(call include_mk, python-package.mk)
 
define Package/python-uwsgi
  SECTION:=language-python
  CATEGORY:=Languages
  SUBMENU:=Python
  TITLE:=The uWSGI server
  URL:=http://unbit.com/
  DEPENDS:=+python +libopenssl +libpcre
endef
 
define Package/python-uwsgi/description
 The uWSGI project 
endef
 
define Build/Compile
	$(call Build/Compile/PyMod,., \
		install --prefix="/usr" --root="$(PKG_INSTALL_DIR)" \
	)
endef
 
define Package/python-uwsgi/install
	$(INSTALL_DIR) $(1)$(PYTHON_PKG_DIR)
	$(INSTALL_DIR) $(1)/usr/bin
	$(CP) \
		$(PKG_INSTALL_DIR)$(PYTHON_PKG_DIR)/* \
		$(1)$(PYTHON_PKG_DIR)/
	$(CP) \
		$(PKG_INSTALL_DIR)/usr/bin/* \
		$(1)/usr/bin/
endef
 
$(eval $(call PyPackage,python-uwsgi))
$(eval $(call BuildPackage,python-uwsgi))
```

but i get 

```
*** uWSGI compiling server core ***
[thread 0][i486-openwrt-linux-musl-gcc] core/utils.o
[thread 3][i486-openwrt-linux-musl-gcc] core/protocol.o
[thread 1][i486-openwrt-linux-musl-gcc] core/socket.o
[thread 2][i486-openwrt-linux-musl-gcc] core/logging.o
core/logging.c: In function 'uwsgi_log':
core/logging.c:51:4: error: implicit declaration of function 'ctime_r' [-Werror=implicit-function-declaration]
    ctime_r((const time_t *) &tv.tv_sec, ctime_storage);
    ^
core/utils.c: In function 'uwsgi_as_root':
core/utils.c:344:7: error: implicit declaration of function 'unshare' [-Werror=implicit-function-declaration]
   if (unshare(uwsgi.unshare)) {
       ^
core/utils.c:564:5: error: implicit declaration of function 'sigfillset' [-Werror=implicit-function-declaration]
     sigfillset(&smask);
     ^
```

have anyone added support for the uwsgi package?

Thanks in advance!

yousong, 2016-08-11:
Maybe it's just `core/logging.c` misses a include `time.h` header file.

diizzyy, 2017-06-09:
@simmyg89 
Please close this and use the forums instead (LEDE or OpenWRT, although LEDE is probably a better idea).

closed 2017-06-09 by hnyman