openwrt/packages #2417: unable to start adblock0.8,without any log file that it should

likemelikeit, 2016-02-24:
Pelesa, i am not a programer, forgivemy low discribe.
Here's my system:
	HuaWei HG255D 
       Firmware Version	OpenWrt Designated Driver r48753 / LuCI Master (git-16.053.73385-1f4f828)
       Kernel Version	4.3.4
Even enable adblock in LUCI config interface, it can't work! (also enable Log options).And the terminal show me this:

root@OpenWrt:~# /etc/init.d/adblock stop
root@OpenWrt:~# /etc/init.d/adblock start
adblock[24733] info : domain adblock processing started (0.80.1, r48753, 24.02.2016 12:52:41)
adblock[24733] error: not enough space in '/tmp/tmp.LgDKiE', please supersize your temp directory, rc: 140
adblock[24733] info : domain adblock processing failed (0.80.1, r48753, 24.02.2016 12:52:42)
root@OpenWrt:~#

the important thing is my /tmp path have enought space,and i can't find the log document where it should be /tmp/adb_debug.log

hnyman, 2016-02-24:
@dibdot

dibdot, 2016-02-24:
@likemelikeit: please ssh to your router and send the output of 'df /tmp' if that happens. Logging will be automatically disabled if your system lacks of enough space, too.

likemelikeit, 2016-02-25:
@hnyman @dibdot: Thanks for you help,i'm make a new sysupgrade.bin and flash it,but adblock still unwork. After running comman 'df /tmp' putty windows show bellow. 
root@OpenWrt:~# df /tmp
Filesystem           1K-blocks      Used Available Use% Mounted on
tmpfs                     14348        568      13780    4%  /tmp
root@OpenWrt:~#

hnyman, 2016-02-25:
> Tue Feb 23 22:42:07 2016 user.notice adblock[2180] error: not enough space in '/tmp/tmp.iGedgl', please supersize your temp directory, rc: 140
>
>
> root@OpenWrt:~# df /tmp
Filesystem 1K-blocks Used Available Use% Mounted on
tmpfs 14348 568 13780 4% /tmp

Then it is likely due to the limited free space, just like the error message says.
The minimum free space requirement in the script is current 20000 kB and you have only 13780 kB.

You might try editing the /usr/bin/adblock-helper.sh script. On line 157, decrease the value adb_minspace=20000 to adb_minspace=10000 and check if the script can successfully run.
```
    # set more script defaults (can't be overwritten by adblock config options)
    #
    adb_cnt=0
    adb_minspace=20000
    adb_unique=1
```
I am not sure if that much (20 MB) free disk space is actually needed with the current script, so it might be possible to run the script with even less free space. (But if the script exhausts disk space, you might encounter problems, even a crash.)

dibdot, 2016-02-25:
@likemelikeit @hnyman , thanks: the original minspace value was intended for processing _all_ blocklist incl. shallalist (the single shallalist archive takes about 10 MB during processing). I'll change the script behaviour for 0.80.2, to print only an info/warning about the low space value.

likemelikeit, 2016-02-25:
@hnyman @dibdot  thanks ,I change the minspace as 10000,now it work! Waitting for next edition.

closed 2016-02-25 by hnyman