openwrt/packages #2421: grilo / lcdgrilo dependencies API

plntyk, 2016-02-25:
lcdgrilo requires grilo-0.2 (checks for that version)
grilo-0.3 has different API: I get a build error if i just change the pkg-config check

reverting 4ceaaaa69604d4034062db193eab7bdcb7819623 seems to fix that
(or updating lcdgrilo)

@MikePetullo

taeasy, 2016-11-04:
make[3]: Entering directory '/home/main/x86/feeds/packages/multimedia/lcdgrilo'
CFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10:lcdgrilo-0.0.10 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/include  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10:lcdgrilo-0.0.10 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/include  -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/include -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include/fortify -I/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/include " LDFLAGS="-L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/usr/lib -L/home/main/x86/staging_dir/toolchain-i386_pentium4_gcc-5.3.0_musl-1.1.15/lib -znow -zrelro -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/lib -L/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/lib " make -j1 -C /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10/. AR="i486-openwrt-linux-musl-gcc-ar" AS="i486-openwrt-linux-musl-gcc -c -Os -pipe -march=pentium4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10:lcdgrilo-0.0.10 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libiconv-stub/include -I/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/usr/lib/libintl-stub/include" LD=i486-openwrt-linux-musl-ld NM="i486-openwrt-linux-musl-gcc-nm" CC="i486-openwrt-linux-musl-gcc" GCC="i486-openwrt-linux-musl-gcc" CXX="i486-openwrt-linux-musl-g++" RANLIB="i486-openwrt-linux-musl-gcc-ranlib" STRIP=i486-openwrt-linux-musl-strip OBJCOPY=i486-openwrt-linux-musl-objcopy OBJDUMP=i486-openwrt-linux-musl-objdump SIZE=i486-openwrt-linux-musl-size CROSS="i486-openwrt-linux-musl-" ARCH="i386" ;
make[4]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10'
make  all-recursive
make[5]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10'
Making all in distro
make[6]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10/distro'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10/distro'
Making all in src
make[6]: Entering directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10/src'
/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/host/bin/valac -c --vapidir=. --pkg posix --pkg pifacecad --pkg gee-0.8 --pkg gstreamer-1.0 --pkg grilo-0.3 --pkg libsoup-2.4 --pkg glib-2.0 --pkg avahi-gobject lcdgrilo.vala stack.vala state.vala state-choose-category.vala state-choose-source.vala state-connecting.vala state-play.vala -C
error: Package `grilo-0.3' not found in specified Vala API directories or GObject-Introspection GIR directories
Compilation failed: 1 error(s), 0 warning(s)
Makefile:710: recipe for target 'lcdgrilo.stamp' failed
make[6]: *** [lcdgrilo.stamp] Error 1
make[6]: Leaving directory '/home/main/x86/build_dir/target-i386_pentium4_musl-1.1.15/lcdgrilo-0.0.10/src'
Makefile:413: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
[... cut ...]
make[2]: *** [package/feeds/packages/lcdgrilo/compile] Error 2
make[2]: Leaving directory '/home/main/x86'
package/Makefile:193: recipe for target '/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/stamp/.package_compile' failed
make[1]: *** [/home/main/x86/staging_dir/target-i386_pentium4_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/main/x86'
/home/main/x86/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
main@debian:~/x86$

dibdot, 2018-08-17:
feel free to open a new ticket, if this is still an issue.

closed 2018-08-17 by dibdot