openwrt/packages #2424: libv4l / v4l-utils / option --set-ctrl seems broken

bittorf, 2016-02-26:
this used to work with older versions
```
root@box:~ v4l2-ctl --set-ctrl focus_auto=0
No value given to suboption <(null)>
[...]
```
(no matter which control i want to set)
i'am on trunk r48773 with musl

thess, 2016-02-26:
It appears that v4l2-ctl in 1.6.3 is broken. Upgrading to the latest version (1.10.0) should fix this problem. I'll let you know when I get it building and test it out.

bittorf, 2016-02-26:
thank you! i'am eager to test it...

thess, 2016-02-27:
Even though the latest version of v4l-utils fixed the option parsing problem, it still did not work correctly with musl (segfault) due to a curious side-effect in getsubopt() found only in glibc. See PR #2430 for a fix/workaround until musl is made compatible with the documented behavior on failure.

closed 2016-02-27 by bittorf