openwrt/packages #2429: mwan3 mark all IPv6 links as down

NainKult, 2016-02-27:
Software versions :

OpenWrt - OpenWrt Designated Driver r48808
LuCI - git-16.057.47821-06cf8b1

mwan3 - 2.0-2
mwan3-luci - 1.4-4

MWAN3 Troubleshooting output: [mwan3_troubleshooting.txt](https://github.com/openwrt/packages/files/149360/mwan3_troubleshooting.txt)
_
Note: Using 6to4, testing with DHCPv6 in a few_

Issue:
As discussed in ​https://forum.openwrt.org/viewtopic.php?pid=312654#p312654
mwan3 seems to get a wrong gateway from lib/functions/network.sh (at least in 6to4)

mwan3 also report that no default gateway is present in the main routing table (which is wrong, see included route -n)

Output from script used by mwan3:
```
root@net002:~# . /lib/functions/network.sh && network_get_gateway6 gateway WAN2_V6 && echo $gateway
::
root@net002:~# . /lib/functions/network.sh && network_get_gateway gateway WAN2 && echo $gateway
*.*.252.1
```

Can't ping the gateway returned from the script:
```
root@net002:~# ping6 -I 6in4-WAN2_V6 ::
PING :: (::): 56 data bytes
ping6: sendto: Permission denied
root@net002:~# ping6 -I 6in4-WAN2_V6 ::1
PING ::1 (::1): 56 data bytes
ping6: sendto: Permission denied
```

Can ping the monitoring IP:
```
root@net002:~# ping6 -I 6in4-WAN2_V6 2001:4860:4860::8888
PING 2001:4860:4860::8888 (2001:4860:4860::8888): 56 data bytes
64 bytes from 2001:4860:4860::8888: seq=0 ttl=54 time=25.668 ms
64 bytes from 2001:4860:4860::8888: seq=1 ttl=54 time=25.678 ms
^C
--- 2001:4860:4860::8888 ping statistics ---
2 packets transmitted, 2 packets received, 0% packet loss
round-trip min/avg/max = 25.668/25.673/25.678 ms
```

NainKult, 2016-02-27:
@arfett @Adze1502

arfett, 2016-02-27:
For the time being the GUI for mwan3 is really only built with IPv4 in mind. You can probably disregard error messages (from the GUI) for the time being. When IPv6 in mwan3 is more fleshed out and adze tells me how everything should be working there will be a new update to the GUI.

NainKult, 2016-02-27:
Copy that, thanks for the heads up.
Ill try to focus on the command-line mwan3 (not the Luci GUI) and reset my Luci generated config files.

What my actual mwan3 report:
```
Directly connected ipv6 networks:
 2001:470:****:1::/64
 fe80::/64
 2001:470:****:2::/64
 2001:470:****:3::/64
```
-> 6to4 tunnel and routed prefix doesn't show up

My 6to4 routes where the /64 is the tunnel endpoint and /48 my routed prefix.
```
root@net002:~# ip -6 route show
default from 2001:470:****:****::/64 dev 6in4-WAN2_V6  proto static  metric 35  pref medium
default from 2001:470:****::/48 dev 6in4-WAN2_V6  proto static  metric 35  pref medium
```

I will test more things up in a few (with DHCPv6 or Static IPv6) cause i think its caused by mwan3 not being able to recognize 6to4.

_I'm an IPv6 enthusiast but i'm still bound to 6to4 cause my freakin' ISP doesn't seems to be moving that fast to adopt full-stack IPv6. Let me know if you need more informations or you want me to test things for you._

Thanks for your work.

yousong, 2016-08-13:
Any news on this one?

arfett, 2016-08-22:
I haven't really been following any news with mwan3 and haven't heard from Adze in a while. Again if I have good documentation on how IPv6 works with mwan3 and a clear direction to take I can update mwan3 GUI over the next few weeks. I haven't studied up on IPv6 in some time and don't really use it at work or home since I'm in the USA and IPv6 tends to break things. I could possibly use some help from the IPv6 enthusiasts or Adze on how things work and how people want them to appear on the GUI.

yousong, 2016-08-22:
Then it's a patches-are-welcome state ;)

Schimmelreiter, 2016-09-07:
For my 6in4 WAN6, i changed
/lib/mwan3/mwan3.sh

I found the line
`IPT6="/usr/sbin/ip6tables -t mangle -w"`
and changed to
`#IPT6="/usr/sbin/ip6tables -t mangle -w"`
`IPT6="/bin/true"`

For normal aiccu/6in4/6to4 wan6 connections, mwan3 should do nothing:
There aren't multiple wan6 connections to chose from but only one tunnel.
If mwan3 does its IPv4 job right the tunnel will re-appear on second or third IPv4-wan if the first/second goes down.

In case we **really** have multiple IP(v6) capable wan connections, simply switching the gateway is not enough, we would also need to modify all addresses when switching interfaces.
Announcing only an ULA prefix in LAN and replacing the ULA prefix with the interface's corresponding public prefix (and replacing our public prefixes with our ULA for incoming traffic ...) could be a solution, but I do not think mwan3 handles this yet, so just don't have mwan3 mess around with IPv6 at all for the moment.

dersch81, 2017-11-01:
Why is it closed? That issue is still not fixed.

NainKult, 2017-11-02:
I tend to close issues opened by me without activity for at least one year. If you want to keep the thread alive, be my guest :)

Per9, 2018-07-22:
The newest version in master seems to work with ipv6 now, but you need to enable the wan6 interface in mwan3.

dibdot, 2018-08-18:
seems to be fixed.

closed 2018-08-18 by dibdot

gaia, 2018-08-30:
I'm using 18.06.1 and after installing mwan3 IPv6 (via HE) stops working on the router and on the clients.

Enabling wan6 in `Switch` (on LuCI) enables IPv6 on the router but still not on the clients, even after a network restart by the client.

I would like to at least be able to install mwan3 and have it co-exist with the 6in4 tunnel even when another LAN is not available.

So my question is: @dibdot what exactly is fixed?

Per9, 2018-09-14:
@gaia : It worked with this commit: https://github.com/openwrt/packages/commit/b6249f1781efc4fefbdf87b661d53c0923ec7438#diff-34590d11545c84b961d1992ce3aefdeb

I think 18.06.1 still contains the version before this commit, so if you download and replace https://github.com/openwrt/packages/blob/b6249f1781efc4fefbdf87b661d53c0923ec7438/net/mwan3/files/lib/mwan3/mwan3.sh on your router ipv6 may work.