openwrt/packages #2440: node-cyclon fails to build if npm is not installed on the host

zoobab, 2016-02-29:
Hi,

I have this build failure for node-cyclon (gpio support select as a module), it fails to find npm:

```
touch /home/zoobab/soft/openwrt/staging_dir/target-mips_34kc_musl-1.1.12/root-ar71xx/stamp/.node_installed
make[3]: Leaving directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node'
make[3]: Entering directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node-serialport'
make[3]: Leaving directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node-serialport'
make[3]: Entering directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node-arduino-firmata'
make[3]: Leaving directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node-arduino-firmata'
make[3]: Entering directory '/home/zoobab/soft/openwrt/feeds/packages/lang/node-cylon'
AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-cylon-0.22.0:node-cylon-0.22.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" npm_config_arch="mips" npm_config_nodedir=/home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-v0.12.7/ PREFIX="/home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-cylon-0.22.0/ipkg-install/usr/" /home/zoobab/soft/openwrt/staging_dir/target-mips_34kc_musl-1.1.12/host/bin/npm install -g /home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-cylon-0.22.0
bash: /home/zoobab/soft/openwrt/staging_dir/target-mips_34kc_musl-1.1.12/host/bin/npm: No such file or directory
Makefile:94: recipe for target '/home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-cylon-0.22.0/.built' failed
make[3]: *** [/home/zoobab/soft/openwrt/build_dir/target-mips_34kc_musl-1.1.12/node-cylon-0.22.0/.built] Error 127
```

Does it needs a host dependency towards npm?

nxhack, 2016-03-11:
+$(STAGING_DIR)/root-$(BOARD)/usr/bin/npm install -g $(PKG_BUILD_DIR)

nxhack, 2016-03-11:
Makefile

-       $(STAGING_DIR)/host/bin/npm install -g $(PKG_BUILD_DIR)
+       $(STAGING_DIR)/root-$(BOARD)/usr/bin/npm install -g $(PKG_BUILD_DIR)

stintel, 2016-03-17:
@nxhack that doesn't work if you don't have node installed on the host:
```bash: /home/LENIO/build/OpenWrt/git/staging_dir/target-x86_64_musl-1.1.14/host/bin/npm: No such file or directory```

This problem exists for other node-* packages as well.
@blogic you are maintainer, do you mind having a look?

blogic, 2016-03-17:
> @nxhack <https://github.com/nxhack> that doesn't work if you don't have
> node installed on the host:
> |bash:
> /home/LENIO/build/OpenWrt/git/staging_dir/target-x86_64_musl-1.1.14/host/bin/npm:
> No such [...]

i'll add a host dep for npm later on today busy with other stuff just now

PR #2513 (2016-03-19): Add node host dependency to node-* packages
Fixes #2440.
closed 2016-03-19 by blogic