openwrt/packages #2442: node crash on mips (AR9330)

zoobab, 2016-02-29:
Hi,

Node is crashing when I try to do a simple hello world on MIPS:

```
root@OpenWrt:/tmp# cat /root/hello-nodejs.js 
console.log("Hello World");
root@OpenWrt:/tmp# node /root/hello-nodejs.js 
Illegal instruction
root@OpenWrt:/tmp# node -v
v0.12.7
```

I am building it on a r48724, and compiling the same and executing it on x64 target works fine:

```
~ # node -v
v0.12.7
~ # node /root/hello.js 
Hello World
```

tpham3783, 2016-02-29:
​Ben,

Can you make sure software floating point is supported in the kernel.  I
got the same error the other day and enabling softfloat had solved it.

-tp

tpham3783, 2016-02-29:
​Check to see if this config is enabled while building your target firmware:

CONFIG_SOFT_FLOAT=y

zoobab, 2016-02-29:
Yes, it is enabled:

zoobab@esivisu /home/zoobab/soft/openwrt [16]$ grep "CONFIG_SOFT_FLOAT" .config
CONFIG_SOFT_FLOAT=y

I could quickly check if I have the same problem with nodejs+15.05.

tpham3783, 2016-02-29:
​Please make sure that this option is enabled in the kernel too, its under
kernel type->MIPS FPU Emulation.

CONFIG_MIPS_FPU_EMULATOR=y

zoobab, 2016-02-29:
Yeah, now that works. Is it necessary to change the node package for this problem?

tpham3783, 2016-02-29:
​Great!

It is actually a target/board config.  Some targets will support hardfloat,
some will support softfloat, others by default wont support any.  Its a
good idea to enable softfloat emulation by default for all targets if no
hard-float is detected in the target CPU.  I'll leave it up to the
developers as to how to enable the feature.

karlp, 2016-05-24:
Duplicate of https://github.com/openwrt/packages/issues/2633 ?

diizzyy, 2017-06-02:
@zoobab 
Is this still an issue or is it fixed?

diizzyy, 2017-06-09:
Problem/Solution explained in #2633 
@thess @hnyman @NeoRaider 
Please close this

closed 2018-01-26 by yousong