openwrt/packages #2445: ddns bug again to freedns.afraid.org - not require "user" (only password from sites scripts)

build000, 2016-03-01:
> 
(...)
051217       : Update needed - L: '1.2.3.4' <> R: '1.2.3.3'
 051217       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/orange_afraid.dat -o /var/run/ddns/orange_afraid.err --no-proxy 'https://xxxxxxxx:yyyyyyyyyyyyyyyyyyyyyyyyyyyyyyy@freedns.afraid.org/nic/update?hostname=aaa.bbb.ccc&myip=1.2.3.4'
 051219       : DDNS Provider answered:
ERROR: Unable to locate this record (changed password recently? deleted and re-created this dns entry?) (double check username/password are correct)
 051219  info : Update successful - IP '1.2.3.4' send
 051219  info : Forced update successful - IP: '1.2.3.4' send
 051219       : Waiting 600 seconds (Check Interval)

Effects: not effectively changes IP adress (visible to LuCI)
Of course, 'user' and 'password' enter the correct, consistent with the registration freedns.afraid.org.
If not use ssl - problem is identically.
Of course the data are examples to show the essence of the problem: xxxxx is the 'user' and 'er ...' is the domain name.

chris5560, 2016-03-02:
It's always a discussion with http://freedns.afraid.org
I have no account at freedns.afraid.org so I cannot test.
Please provide me (possibly per direct mail) your "Direct URL" from freedns.afraid.org and please give me a comment what are the data inside this URL.
So I can edit services file inside ddns-scripts to the right URL.
Thanks for your support.

build000, 2016-03-03:
@chris5560
So create an account there and it himself Investigate - the service is free - after half a year just gone.
;)

diizzyy, 2016-03-06:
Why is this reverted when it's working?

`
" 230620       : Forced Update - L: '23.255.99.111' == R: ''23.255.99.111'
 230620       : #> /usr/bin/curl -RsS -o /var/run/ddns/freedns.dat --stderr /var/run/ddns/freedns.err --noproxy '*' 'http://username:passwd@freedns.afraid.org/nic/update?hostname=hostname.somewhere.org&myip='23.255.99.111'
 230620       : DDNS Provider answered:
ERROR: Address '23.255.99.111 has not changed.
 230620  info : Forced update successful - IP: ''23.255.99.111' send
 230621       : Waiting 600 seconds (Check Interval)
`

`ERROR: Unable to locate this record (changed password recently? deleted and re-created this dns entry?) (double check username/password are correct)` 
That's a configuration issue on the user behalf...

Here are a few more sources verifyingthat the URL is indeed correct:
https://www.reddit.com/r/webdev/comments/3xeteh/is_there_any_free_dns_services_that_allow/cy41h5y
http://quabr.com/35203039/ddns-update-key-in-python-sending-username-password-domainname-and-ip

chris5560, 2016-03-07:
@diizzyy
Could you provide the documentation of freedns.afraid.org where your used update_url is documented.
Every where I googled I only found the currently used url. It's also used and documented for IPv6 usage.

diizzyy, 2016-03-07:
@chris5560 

If you login, go to  http://freedns.afraid.org/dynamic/

```
You can also (alternatively) perform updates like this:

http://[USERNAME]:[PASSWORD]@freedns.afraid.org/nic/update?hostname=[DOMAIN]&myip=[IP]
(https/SSL is also optionally supported/available)
```

The direct links on the same page uses the the API key which doesn't really line up with current configuration.
http://freedns.afraid.org/dynamic/update.php?....

chris5560, 2016-03-07:
@diizzyy
For the moment I prefer public documentation at http://freedns.afraid.org/scripts/freedns.clients.php working for years with the existing URL. As seen in this issue here, your suggested url breaks existing users configuration.

diizzyy, 2016-03-07:
My change actually fixes the incorrect setup (API key is as I said not password which is what you actually label that box), if you want to avoid POLA (https://www.freebsd.org/doc/handbook/freebsd-glossary.html#pola-glossary) at least add the other URL as an additional option which actually carries the correct variables.

chris5560, 2016-03-07:
As described at freedns.afraid.org fill in the "token" into "password" field.
That's what done and working since OpenWrt 10.03
Other router software say for using freedns.afraid.org: set ddns service to "custom" and copy the given url into "custom url".
This also works for ddns-scripts, if you prefere.
There are other providers also using "token" that are filled into "password" field.
services- and services_ipv6-files are hidden to casual users.
If the provider say fill "x" into field "y" when using OpenWrt, then it should work this way. So has nothing to do with POLA.
How may different fields should ddns-scripts handle to fulfil every provider needs, who is not using DynDNS API.

P.S. please provide PUBLIC access to freedns.afraid.org API. I will have a look at.

diizzyy, 2016-03-07:
..and I'm telling you that wording is wrong. Anyhow to access the API documentation you need to have an account for this service, feel free to voice your concerns to the site admin about that.
I did however have a look at ddclient's source and they're using that "old" URL but actually parsing the correct login data. https://github.com/wimpunk/ddclient/blob/master/ddclient#L3761
If that feels more confortable please add that but I still insist that user/pass should correspond to user/pass, just because you've been doing something "wrong" for the last years doesn't justify it. At least provide a way to input that correct login data as an option.

build000, 2016-03-08:
https://github.com/chris5560/openwrt-packages/commit/38866917d85e513436b61be464c07f8454f3edac - confirm: fix this bug

closed 2016-03-08 by build000