openwrt/packages #2447: vnstat is sometimes adding 4GB of data to statistic when running command "wifi"

Misiek304, 2016-03-01:
I have set my router to reset wlan0 every 6 hours using cron and wifi command.
Sometimes after running that command vnstat is adding 4GB of transfer to my wlan0 statistic. Anyone know why?

hnyman, 2016-03-01:
Likely a logic where vnstat tries to automatically correspond to a 32 bit integer's overflow. Vnstat has likely a 64 bit internal traffic counter and assumes that the reseted traffic counter (smaller cumulative traffic than at the earlier stats collection) for the interface means that the interface's traffic counter is 32bit and has overflowed at 2^32 bytes of traffic = ~4 GB. As you say "sometimes", I guess that vnstat developers have built a threshold to trigger that behaviour, e.g. the previous traffic counter needs to have been over 2 GB (2^31) or something like that to trigger the + 4 GB behaviour.

Sounds like something you might report to vnstat developers to be fixed there. (but that might actually be a purposeful smart feature from their perspective.)

Misiek304, 2016-03-01:
I also noticed that there is available vnstat 1.15 version but in OpenWRT repo there is only 1.12. Could this be updated?

diizzyy, 2017-06-09:
@Misiek304 
New version of vnstat(d) is in the repo, please test with LEDE and report upstream if this is still an issue.

@hnyman 
Please close this.

closed 2017-06-10 by Misiek304