openwrt/packages #2448: Strongswan patch 101-musl-fixes.path causing compile failure on ARM platform with 3.4 linux header files

myabacus, 2016-03-02:
In 101-musl-fixes.patch, __kernel_nlink_t is defined as void.

However, in linux3.4 arch/arm/include/asm/posix_types.h, __kernel_nlink_t is defined as:

`typedef unsigned short		__kernel_nlink_t;`
`#define __kernel_nlink_t __kernel_nlink_t`

So these two duplicated defines cause compile failures.

@sbyx

dedeckeh, 2016-10-30:
# #2532 
PR is pending for a while to fix this issue ...

thess, 2016-10-30:
It appears that @sbyx has been MIA for awhile now. If you or someone else were to submit a tested PR for this, I will consider merging it.

dedeckeh, 2016-10-30:
@thess I have hit the same issue so I can launch a tested PR; better to base my PR on version 5.5.1 as in the PR #3357 by @stintel.
Would be great if some other Strongswan PR get also handled; #3039 fixes a compilation issue depending on the used gcc version while #3032 fixes an ip package dependency issue

thess, 2016-10-31:
Without Steven to at least ACK the PR's, this package needs a new maintainer - preferably someone who makes use of the package and will test it. Users/Volunteers?

stintel, 2016-10-31:
@myabacus, @dedeckeh is this still a problem with the current strongswan version? I have just compiled strongswan 5.5.1 for imx6 (arm_cortex-a9_neon) without issue.

dedeckeh, 2016-11-02:
@stintel Confirmation this is not anymore an issue after upstepping to strongswan 5.5.1. So this issue can be closed while PR #2532 can also be closed

stintel, 2016-11-02:
@dedeckeh thanks for the feedback. Closing.

closed 2016-11-02 by stintel

dedeckeh, 2016-11-09:
@stintel I did my test on a wrong target as the problem is still present using a pre-3.6 kernel; an arm cortex and uclibc as clib. Meanwhile I openend PR #3501 to fix the issue; tested it on Lede trunk and brcm63xx target using 3.4 kernel.  Sorry for the initial wrong conclusion