openwrt/packages #2467: Make package/vim/compile fails

auszone, 2016-03-07:
```
checking --with-tlib argument... ncurses
checking for linking with ncurses library... configure: error: FAILED
make[2]: *** [/Volumes/OpenWrt/openwrt/build_dir/target-mipsel_24kec+dsp_glibc-2.21/vim74/.configured_yyynnnnnn] Error 1
make[2]: Leaving directory `/Volumes/OpenWrt/openwrt/feeds/packages/utils/vim'
make[1]: *** [package/feeds/packages/vim/compile] Error 2
make[1]: Leaving directory `/Volumes/OpenWrt/openwrt'
make: *** [package/vim/compile] Error 2
```
The error above happens when I'm trying to build vim on the host.
However, I can build ncurses with no problem with `make package/ncurses/compile`.
Also, I can also build nano which also depends on this lib by `make package/nano/compile`.

dfc-d, 2016-03-13:
This worked for me today for ar71xx (that is vim build worked).

auszone, 2016-03-13:
I was at RT288x

auszone, 2016-03-17:
Solved, it's because I was in the OSX environment.

closed 2016-03-17 by auszone