openwrt/packages #2477: minidlna package wont compile

dav01it, 2016-03-09:
Asking if someone else has same issue:
scenario:
fresh git update

compile for AMD GEODE arch (alixboard)
if I add minidlna it stops compiling with error:**_ libFLAC.so: undefined reference to `__sigemptyset'_**
**removing minidlna package from menuconfig I get successfull compilation.
**am I missing some deps ?

dav01it, 2016-03-09:
Found!
http://lists.busybox.net/pipermail/buildroot/2015-April/124949.html
had to edit/patch openwrt/build_dir/target-i386_geode_musl-1.1.14/flac-1.3.1/src/libFLAC/cpu.c
as described in the link.

If someone knows how to submit the patch for everyone please tell me how!

thess, 2016-03-09:
Thanks for finding the fix - I'll get it merged later today

thess, 2016-03-09:
Patch merged 1263599f96f13f11d719ce336dfb6a639b32de98 to libs/flac

closed 2016-03-09 by thess