openwrt/packages #2484: ddns-scripts: username and password leaked for namecheap while it could be avoided

lelutin, 2016-03-10:
The update request for namecheap.com is made over HTTP with all parameters in the URL. this makes it possible for any network listener to grab credentials.

the namecheap documentation gives examples using https:// urls:

https://www.namecheap.com/support/knowledgebase/article.aspx/29/11/how-do-i-use-the-browser-to-dynamically-update-hosts-ip

So I suggest to change the URL to make it use https:// instead (in file "services").

I've made this change locally (on a 15.05 install) and the update works well.

hnyman, 2016-03-10:
@chris5560

jow-, 2016-03-10:
Please keep in mind that there is no default ssl support in wget.

chris5560, 2016-03-10:
@lelutin

SSL(https) is supported since OpenWrt 10.03.

Inside Wiki there are descriptions how to use and configure SSL inside ddns-scripts.

Installation: https://wiki.openwrt.org/doc/howto/ddns.client#ssl_support
Config via LuCI: https://wiki.openwrt.org/doc/howto/ddns.client#ssl_support1
Config via console: https://wiki.openwrt.org/doc/howto/ddns.client#ssl_support2

So there is no need to change url inside services/services_ipv6 files.

If you change url inside services* files you still need to install Wget or Curl and ca-certificates to enable SSL support because BusyBox's Wget does not support SSL.

lelutin, 2016-03-25:
ooh I see. thanks for pointing this out to me. I installed wget and the option was easy to find in luci.

maybe it'd be interesting to either make wget or curl into a dependency to the ddns-scripts package. or if this is not acceptable maybe add a paragraph in the text on the luci page for "Service->Dynamic DNS" that mentions the lack of encryption and how to get secure connections?

lelutin, 2016-03-25:
... or another option is to avoid hiding the "Use HTTP Secure" option in the interface when wget or curl are not installed, make the check box greyed out (disabled) and have a message that says "to enable HTTP Secure, you need to install wget or curl"

chris5560, 2016-03-25:
Gabriel,

when using LuCI (since BB 15.05) there are "Hints" on the Overview page pointing out that there is no SSL support installed incl. telling you to install wget or curl package if ssl is needed.
Additionally the "Enable HTTPS" checkbox is only shown if SSL support is available.

Christian

lelutin, 2016-03-28:
@chris5560 ah.. well I guess I missed the hints about lack of ssl support. imho it would make things clearer if "Enable HTTPS" was always visible but greyed out. but since you indicated there's already hints about missing ssl support, I'll just close this issue.

closed 2016-03-28 by lelutin