openwrt/packages #2506: Mysql update

TeHashX, 2016-03-15:
Hi, can you please update mysql to a more recent version like 5.7?
Thanks

jow-, 2016-03-15:
I won't have time to work on this. Patches welcome.

closed 2016-08-11 by yousong

TeHashX, 2017-07-24:
Hi, trying to install mariadb-server v5.5.56-1 on RT-AC68U router through entware-ng.arm, kernel  v2.6.36.4brcmarm, firmware v380.67 RMerlin.
````
admin@RT-AC68U:/tmp/mnt/sda1# mysql_install_db --force
/opt/bin/mysql_install_db: line 340: hostname: not found
chown: unknown user root
Cannot change ownership of the database directories to the 'root'
user.  Check that you have the necessary permissions and try again.
````
I change user from root to admin in my.cnf and solve hostname
````
cat >> /opt/bin/hostname << 'EOF'
cat /proc/sys/kernel/hostname
EOF
chmod 0755 /opt/bin/hostname
````
Run again
````
admin@RT-AC68U:/tmp/mnt/sda1# mysql_install_db --force
Installing MariaDB/MySQL system tables in '/opt/var/mysql/' ...
170724 15:14:55 [Warning] 'THREAD_CONCURRENCY' is deprecated and will be removed in a future release.
170724 15:14:55 [Note] /opt/bin/mysqld (mysqld 5.5.56-MariaDB) starting as process 8109 ...
OK
Filling help tables...
170724 15:14:56 [Warning] 'THREAD_CONCURRENCY' is deprecated and will be removed in a future release.
170724 15:14:56 [Note] /opt/bin/mysqld (mysqld 5.5.56-MariaDB) starting as process 8122 ...
OK

To start mysqld at boot time you have to copy
support-files/mysql.server to the right place for your system

PLEASE REMEMBER TO SET A PASSWORD FOR THE MariaDB root USER !
To do so, start the server, then issue the following commands:

'/opt/bin/mysqladmin' -u root password 'new-password'
'/opt/bin/mysqladmin' -u root -h RT-AC68U password 'new-password'

Alternatively you can run:
'/opt/bin/mysql_secure_installation'

which will also give you the option of removing the test
databases and anonymous user created by default.  This is
strongly recommended for production servers.

See the MariaDB Knowledgebase at http://mariadb.com/kb or the
MySQL manual for more instructions.

You can start the MariaDB daemon with:
cd '/opt' ; /opt/bin/mysqld_safe --datadir='/opt/var/mysql/'

You can test the MariaDB daemon with mysql-test-run.pl
cd '/opt/mysql-test' ; perl mysql-test-run.pl

Please report any problems at http://mariadb.org/jira

The latest information about MariaDB is available at http://mariadb.org/.
You can find additional information about the MySQL part at:
http://dev.mysql.com
Consider joining MariaDB's strong and vibrant community:
https://mariadb.org/get-involved/
````
Trying to create a password
````
admin@RT-AC68U:/tmp/mnt/sda1# /opt/bin/mysqladmin -u root password new-password
-sh: /opt/bin/mysqladmin: not found
````
List /opt/bin files
````
admin@RT-AC68U:/tmp/mnt/sda1# ls -l /opt/bin
-rwxr-xr-x    1 admin    root       3469440 May 30 20:25 aria_chk
-rwxr-xr-x    1 admin    root       3304184 May 30 20:25 aria_dump_log
[... 76 lines cut ...]
cp /opt/share/mysql/english/errmsg.sys /opt/share/
````
Again
````
admin@RT-AC68U:/tmp/mnt/sda1# /opt/bin/mysql_secure_installation
/opt/bin/mysql_secure_installation: line 393: find_mysql_client: not found

NOTE: RUNNING ALL PARTS OF THIS SCRIPT IS RECOMMENDED FOR ALL MariaDB
      SERVERS IN PRODUCTION USE!  PLEASE READ EACH STEP CAREFULLY!

In order to log into MariaDB to secure it, we'll need the current
password for the root user.  If you've just installed MariaDB, and
you haven't set the root password yet, the password will be blank,
so you should just press enter here.

Enter current password for root (enter for none):
170724 15:22:38 [Note] /opt/bin/mysql (mysqld 5.5.56-MariaDB) starting as process 9312 ...
170724 15:22:38 [Warning] An old style --language or -lc-message-dir value with language specific part detected: /tmp/mnt/sda1/entware-ng.arm/share/
170724 15:22:38 [Warning] Use --lc-messages-dir without language specific part instead.
/opt/bin/mysql: Please consult the Knowledge Base to find out how to run mysqld as root!
170724 15:22:38 [ERROR] Aborting

170724 15:22:38 [Note] /opt/bin/mysql: Shutdown complete
````
Some hint please?

jow-, 2017-07-24:
The package has only been tested with LEDE ad OpenWrt. It is also not prepared to be installed into `/opt`.

tony-sunjian, 2017-09-06:
@TeHashX i have the same problem with you!  have you figure out now？ My router is N-18U

p3x-robot, 2017-10-20:
https://github.com/patrikx3/lede-mariadb 5.5

Chaaang, 2017-11-29:
@jow can you please update mysql server? or just try to use mariadb.

rechinu88, 2018-08-21:
Same problem as TeHashX. There is no mysqladmin file

KYegres, 2018-11-14:
I solved same problem with follow commands:
```
opkg install mariadb-client
mysql -u root
UPDATE mysql.user SET Password=PASSWORD('NEW_PASSWORD') WHERE User='root';
FLUSH PRIVILEGES;
```
Now it works.

After it I found that package mariadb-client-extra contains file mysqladmin :) I didn't try it.