openwrt/packages #2507: Zabbix agentd missing dependencie

gilfrade, 2016-03-15:
Package zabbix-agentd is missing dependencies for the following libraries:
libiconv.so.2
Makefile:242: recipe for target '/home/openwrt/openwrt_cp/bin/apu/packages/packages/zabbix-agentd_2.4.6-2_x86_64.ipk' failed

I compiled/installed libiconv and added the following to zabbix Makefile:

define Package/zabbix-agentd
  $(call Package/zabbix/Default)
  TITLE+= agentd
  DEPENDS += +libiconv
endef

It still complains the missing dependency...

champtar, 2016-03-15:
Hi,
Please add as much info as possible like openwrt revision and your .config file

gilfrade, 2016-03-15:
As requested i am using the following:

git-svn-id: svn://svn.openwrt.org/openwrt/trunk@47516 3c298f89-4303-0410-b956-a3cf2f4a3e73

see config annexed 
[config.txt](https://github.com/openwrt/packages/files/174115/config.txt)

gilfrade, 2016-03-16:
Fixed Makefile change +libiconv to +libiconv-full and now it works

define Package/zabbix-agentd
$(call Package/zabbix/Default)
TITLE+= agentd
DEPENDS += +libiconv-full
endef

champtar, 2016-03-16:
don't close it's not fixed

Noltari, 2016-03-16:
The dependency should be $(ICONV_DEPENDS), not libiconv or libiconv-full:
https://github.com/openwrt/openwrt/blob/master/include/nls.mk

Example: https://github.com/openwrt/packages/commit/8246c8d9b83f0e88d59c16c06734c6009587f0d7

gilfrade, 2016-03-16:
Tested with Noltari solution and works has expected.

define Package/zabbix-agentd
$(call Package/zabbix/Default)
TITLE+= agentd
DEPENDS += $(ICONV_DEPENDS)
endef

champtar, 2016-03-16:
I think i will disable iconv instead of depending on it

Noltari, 2016-03-16:
@champtar I don't see any problem in depending on libiconv, a minimal version of libiconv it's built-in by default in OpenWrt. The problem only appears when full language support is used, due to the missing dependency.

champtar, 2016-03-17:
@Noltari you are right, zabbix do depend on iconv.
All zabbix bin should depend on iconv

commit 8dfc434b (2016-03-22): zabbix: add missing dependency on libiconv ($(ICONV_DEPENDS))

this fixes #2507
thanks @gilfrade and @Noltari
commit e4bfdb14 (2016-03-24): zabbix: add missing dependency on libiconv ($(ICONV_DEPENDS))

this fixes #2507
thanks @gilfrade and @Noltari
closed 2016-03-22 by champtar