openwrt/packages #2511: python-cffi: failing to build for netlogic and octeon

jefferyto, 2016-03-16:
http://buildbot.openwrt.org:8010/broken_packages/netlogic/python-cffi/compile.txt
http://buildbot.openwrt.org:8010/broken_packages/octeon/python-cffi/compile.txt

hnyman, 2016-03-16:
That seems to originate to libffi itself.
python-cffi logs say:
``` 
Package libffi was not found in the pkg-config search path.
Perhaps you should add the directory containing `libffi.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libffi' found
```

Looks like this is not a python-specific problem, but instead the underlying libffi is missing for those two platforms:
http://buildbot.openwrt.org:8010/broken_packages/octeon/libffi/compile.txt
http://buildbot.openwrt.org:8010/broken_packages/netlogic/libffi/compile.txt
```
../src/mips/n32.S: Assembler messages:
../src/mips/n32.S:116: Error: opcode not supported on this processor: mips4 (mips4) `l.s $f12,0*8($25)'
...
../src/mips/n32.S:511: Error: opcode not supported on this processor: mips4 (mips4) `l.s $f2,(3*8)($sp)'
../src/mips/n32.S:516: Error: opcode not supported on this processor: mips4 (mips4) `l.s $f0,(2*8)($sp)'
../src/mips/n32.S:517: Error: opcode not supported on this processor: mips4 (mips4) `l.d $f2,(3*8)($sp)'
make[5]: *** [src/mips/n32.lo] Error 1
```
http://buildbot.openwrt.org:8010/broken_packages/netlogic/libffi/
http://buildbot.openwrt.org:8010/broken_packages/octeon/libffi/

I don't know if libffi can be fixed to compile for those platforms, or maybe it could marked disabled for those two.
@tripolar

jefferyto, 2016-03-17:
I'm seeing similar error messages when compiling libffi for ar71xx:

```
../src/mips/o32.S: Assembler messages:
../src/mips/o32.S:94: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f12,0*4($sp)'
../src/mips/o32.S:101: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f12,0*4($sp)'
...
../src/mips/o32.S:285: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.s $f0,((14*4)-6*4)($fp)'
../src/mips/o32.S:289: Error: opcode not supported on this processor: mips32r2 (mips32r2) `l.d $f0,((14*4)-6*4)($fp)'
make[6]: *** [src/mips/o32.lo] Error 1
```

I'm pretty sure I was able to build it successfully before - I think a change in the buildroot may be causing this.

jefferyto, 2016-03-17:
These seem to be related / the same issue:
#2111 
atgreen/libffi#187
https://dev.openwrt.org/ticket/20726

yousong, 2016-08-11:
The issue should already be resolved.

thess, 2016-08-11:
@yousong - Not true. I began looking at this problem a week ago and have not good solution yet. Perhaps someone more familiar with compiler / arch / cpu options could shed some light.

http://downloads.lede-project.org/snapshots/faillogs/mips64_xlp/packages/libffi/compile.txt

yousong, 2016-08-12:
Hmm, I thought the `libffi/patches/100-fix_mips_softfloat.patch` already fixed it, but it looks like it only fixes the o32 abi while octeon uses n32?

yousong, 2016-08-15:
I just posted a PR to libffi (https://github.com/libffi/libffi/pull/272) .  It just passed the travis ci check so I guess it's worth a try ;)

yousong, 2016-08-15:
Nah... that travis job only checks i386 arch...

thess, 2016-08-16:
@yousong - I tested your patch to n32.S from https://github.com/libffi/libffi/pull/272

Looks good - fixes MIPS64 builds. Could you push this update to our repo.

closed 2016-08-17 by yousong