openwrt/packages #2515: curl hangs in updating at now-ip.com

willempi, 2016-03-17:
Using openwrt on a Dlink DIR-505.  (ar71xx architecture).

In trunk r49022 and before it is not possible to update an IP number at now-ip.com using their API:
'curl -u user:passwd "https://now-ip.com/update?hostname=your.host.name"'.
With ca-certificates installed or just using cacert.pem  curl just hangs forever.
With strace it always ends like this:

clock_gettime(CLOCK_MONOTONIC, {5516, 402637367}) = 0
clock_gettime(CLOCK_REALTIME, {1458224852, 900731806}) = 0
write(3, "\26\3\1\0\225\1\0\0\221\3\3V\352\276\324>\352\355b\353\364?\347\340@\376\330\331\3735Dg"..., 154) = 154
read(3, 0xa38088, 5)                    = -1 EAGAIN (Resource temporarily unavailable)
fcntl64(3, F_GETFL)                     = 0x82 (flags O_RDWR|O_NONBLOCK)
clock_gettime(CLOCK_MONOTONIC, {5516, 407318638}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 408311456}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 408722927}) = 0
poll([{fd=3, events=POLLIN}], 1, 0)     = 0 (Timeout)
clock_gettime(CLOCK_MONOTONIC, {5516, 411064145}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 411434776}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 411752477}) = 0
poll([{fd=3, events=POLLIN|POLLPRI|POLLRDNORM|POLLRDBAND}], 1, 0) = 0 (Timeout)
clock_gettime(CLOCK_MONOTONIC, {5516, 412688685}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 413169242}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 413558868}) = 0
poll([{fd=3, events=POLLIN}], 1, 1000)  = 1 ([{fd=3, revents=POLLIN}])
clock_gettime(CLOCK_MONOTONIC, {5516, 506441849}) = 0
clock_gettime(CLOCK_MONOTONIC, {5516, 506758525}) = 0
poll([{fd=3, events=POLLIN|POLLPRI|POLLRDNORM|POLLRDBAND}], 1, 0) = 1 ([{fd=3, revents=POLLIN|POLLRDNORM}])
read(3, "\25\3\3\0\2", 5)               = 5
read(3, "\1p", 2)                       = 2

and then nothing ever happens after here.
And the update did not occur.

yousong, 2016-08-11:
Is it possible that it's an ephemeral network issue or issue with now-ip.com's service?

Feel free to re-open this if it persists

closed 2016-08-11 by yousong

willempi, 2016-10-01:
On Wed, Aug 10, 2016 at 07:19:53PM -0700, Yousong Zhou wrote:

That is interesting! The only thing I've ever heard back from github is 'Case closed'.

> Is it possible that it's an ephemeral network issue or issue with now-ip.com's service?

Of coarse a network issue may be possible. It is known that one of our
providers tampers with ssh connections. And of coarse it is possible that it
is a problem at now-ip's side.
But the effect remains the same using different providers, so a network
issue is not likely.
Worse, when I use a 'real machine' behind NAT via that same Dlink device
everything works perfeclty. So it's definitely not a network issue. And it's
not likely that there is a problem at now-ip.
 
Now, gia github was just another attempt to bring it under your attention.
You already knew the problems with polar-ssl as you can see at
https://dev.openwrt.org/ticket/19621 .

Googling around shows nice articles like
http://blog.metamatt.com/blog/2012/03/19/openwrt-is-losing-its-luster
which learns me estimate the speed of resolving bugs after being reported.

Regards, Wim.

yousong, 2016-10-08:
curl is part of base packages (packages maintained along with the build systems etc.)  So you may not get much help here.

The issue may get more attention at the dev mailing lists or OpenWrt ticket system, but mostly it's still a "help yourself" style of support where the issue is more likely to be finally dissected and the culprit pinpointed by those most-affected.

For me, the big thing about OpenWrt/LEDE is the possibilities it provides.  It more or less works: works for me and for a lot of other people most of the time, and I can do the fix when it doesn't.  Really there is no need to feel frustrated.

yousong, 2017-01-17:
In case it may help, a curl upstream patch was recently introduced into LEDE curl package: [curl: fix HTTPS network timeouts with OpenSSL](https://git.lede-project.org/?p=source.git;a=commitdiff;h=b65572fee9d6a31de1a49fe5f2902670e8c86253;hp=b9a408c2b49ccfa0e906bda00ef77f4002e401fd)

willempi, 2017-02-02:
On Mon, Jan 16, 2017 at 11:29:54PM -0800, Yousong Zhou wrote:
> In case it may help, a curl upstream patch was recently introduced into
> LEDE curl package: [curl: fix HTTPS network timeouts with OpenSSL]
> (https://git.lede-project.org/?p=source.git;a=commitdiff;
> h=b65572fee9d6a31de1a49fe5f2902670e8c86253;
> hp=b9a408c2b49ccfa0e906bda00ef77f4002e401fd)

Just found the time to try an upgrade. I had to choose a development
snaphot.  (Lede has the same problem as openwrt had: when I install a
stable branch, then opkg will not install the packages I need.)
Todays snapshot offered me a curl with libmbedtls (instead of polarssl).
And hooray, it works!  That is, after I discovered that now-ip.com renamed
itself recently to now-dns.com, without notification. 

Now I can have a cronjob do things better.
Thanks, Wim.