openwrt/packages #2522: HG255D can't use USB building from trunk source

likemelikeit, 2016-03-19:
The detail of router HG255D is https://wiki.openwrt.org/toh/hwdata/huawei/huawei_hg255d

For it no upgrade file from offical(support hardware), I build it myself,and found that usb did't enable(i'm sure i have installed pakages to do with usb mount and write)

yousong, 2016-03-19:
> For it no upgrade file from offical(support hardware), I build it myself,and found that usb did't enable(i'm sure i have installed pakages to do with usb mount and write)

HG255d is now in unmaintained state.  The image generation code is still there at the code base, but commented out.  See details at https://lists.openwrt.org/pipermail/openwrt-devel/2013-July/020785.html

Please consider opening a ticket at dev.openwrt.org or at the mailing list as the repo here is for packages feed only.

closed 2016-03-19 by yousong

likemelikeit, 2016-03-20:
@yousong thanks for your help！