openwrt/packages #2524: Is raspberry pi kmod-net-rtl8188eu grabbing up the `.ko` from the right path?

cusspvz, 2016-03-19:
I did a fresh install of the OpenWRT snapshot on the RPi.

Everything went well until I tried to install `kmod-net-rtl8188eu`, the package was successfully installed but there was no driver, so I've checked package's contents and it was empty.

[image]

I think path for the driver has changed, could that be the reason?

NOTE: `r8188ru.ko` is placed at `staging` directory after a kernel build.

LipkeGu, 2016-05-23:
there are 2 Drivers for the 81xx:
- rtl8192xx <-- works also for some 8188xx chips (like 8188cu) :)
- rtl8188ru

cusspvz, 2016-05-23:
A further investigation lead me into the patch where `8188eu` was removed because it was only compatible with `wext` driver at the time.

diizzyy, 2017-06-09:
@thess @hnyman 
Fixed in LEDE, please close.

closed 2017-06-09 by thess