openwrt/packages #2529: fwknopd wrong file permission

coelner, 2016-03-21:
the script in rc.d doesn't set the correct file permission.
If you add after line 55 '> /etc/fwknop/access.conf'
```
chmod 600 /etc/fwknop/fwknopd.conf
chmod 600 /etc/fwknop/access.conf
```

the script sets the right permission and the daemon doesn't warn.

oneru, 2016-03-21:
Looks reasonable, I'll take a closer look soon. Thanks!

yousong, 2016-08-11:
@jp-bennett , updates on this?

jp-bennett, 2016-08-12:
@yousong Thanks for the ping, will take care of this this weekend.

jp-bennett, 2016-08-17:
PR3075 contains this fix.

yousong, 2016-08-17:
Thanks.

BTW, I noticed that the init script would write to /etc/fwknopd/ which is very likely on flash storage on devices running OpneWrt/LEDE.  It would be better if it could be moved to /var/etc

closed 2016-08-17 by yousong

jp-bennett, 2016-08-17:
@yousong That's an interesting thought.  I believe that would be possible without too much trouble.  I'll pursue when I get a chance.