openwrt/packages #2531: Flashrom no longer compiles

notnyt, 2016-03-21:
1348057082e5887201f0147fb2eb27d46ef5c6ac broke flashrom.

make[4]: Entering directory '/home/nyt/archive/openwrt/apu2/build_dir/target-x86_64_musl-1.1.14/flashrom-full/flashrom-0.9.9'
Files don't seem to be under version control. Replacing all version templates with 0.9.9-r1955.
Checking for a C compiler... found.
Target arch is x86
Target OS is Linux
Checking for libpci headers... found.
Checking version of pci_get_dev... new version (including PCI domain parameter).
Checking if libpci is present and sufficient... no.
Checking if libz+libpci are present and sufficient...no.

The following features require libpci: CONFIG_INTERNAL CONFIG_NIC3COM CONFIG_GFXNVIDIA CONFIG_SATASII CONFIG_ATAVIA CONFIG_IT8212 CONFIG_DRKAISER CONFIG_NICREALTEK CONFIG_NICINTEL CONFIG_NICINTEL_SPI CONFIG_NICINTEL_EEPROM CONFIG_OGP_SPI CONFIG_SATAMV.
Please install libpci (package pciutils) and/or libz or disable all features
mentioned above by specifying make CONFIG_ENABLE_LIBPCI_PROGRAMMERS=no
See README for more information.

navinp1281, 2016-03-23:
make CONFIG_ENABLE_LIBPCI_PROGRAMMERS=no

along with a CFEATURE_FLAGS += -D'PROGRAMMER_INTERNAL=0'
in both Makefiles of flashrom-pci and flashrom-full fixes the issue.

Noltari, 2016-03-23:
That disables PCI programmers. I know a better way to fix this, but I won't be able to do it until next week.

navinp1281, 2016-03-24:
I was speaking of the target mips . I thought pci was only for x86.

Noltari, 2016-03-28:
Should be fixed now: https://github.com/openwrt/packages/commit/375f34d362ff26feb00ad544c545d46511607914

closed 2016-03-28 by Noltari