openwrt/packages #2548: openconnect / vpnc-script : routes not added

fatchoy, 2016-03-24:
Hello,
I'm trying to setup OpenConnect to route all traffic through the tunnel. I'm using OpenWrt CC 15.05
ocserv is setup with route = 0.0.0.0/0
ocserv seems to be working correctly I can connect from another linux and android client and the default route is setup correctly.
On OpenWrt, I've setup my OpenConnect interface as in the example and the connection is established correctly, however the default route sent by ocserv is not added.
I've tried to use the option defaultroute '1', the connection is established then disconnect immediately. In that case the reason seems that the route to the ocserv gateway is not added, only the default route through openvpn.

nmav, 2016-04-21:
What is the output in the log file?

fatchoy, 2016-04-24:
Nothing special in the logs. 
I've pinpointed the problem to the line 141 in the vpnc-script
the function proto_add_ipv4_route "0.0.0.0" 0 does not work, I've tried added the function out of the loop to exectute it all the time but the default 0.0.0.0/0 route seems to be added only if the option use default route is set to 1.
For the issue about the vpn gateway beeing unreachable, I've used some bits of the original vpnc-script to setup the route to the vpn-gateway ip through the wan interface and to save and restore the default route.

```
DEFAULT_ROUTE_FILE=/var/etc/openconnect-defaultroute

get_default_gw() {
	netstat -r -n | awk '/:/ { next; } /^(default|0\.0\.0\.0)/ { print $2; }'
}

set_vpngateway_route() {
	route add -host "$VPNGATEWAY" gw "`get_default_gw`"
	echo "`get_default_gw`" > "$DEFAULT_ROUTE_FILE"
}

del_vpngateway_route() {
	route del -host "$VPNGATEWAY"
	if [ -s "$DEFAULT_ROUTE_FILE" ]; then
		route add default gw `cat "$DEFAULT_ROUTE_FILE"`
		rm -f -- "$DEFAULT_ROUTE_FILE"
	fi
}
```

nmav, 2016-04-25:
About the default route I don't think the script is right. Could you try replacing the:
```proto_add_ipv4_route "0.0.0.0" 0```
with ```proto_add_ipv4_route "0.0.0.0" 0 $INTERNAL_IP4_ADDRESS```?

nmav, 2016-04-28:
@fatchoy could you try the change above?

fatchoy, 2016-04-28:
@nmav I will do it this weekend when I'm back home, I don't have the router with me at the moment.

fatchoy, 2016-04-28:
@nmav I got my hand on an OpenWrt, I made the change you suggested but no change. 
When option defaultroute is set to 0 the default route is still not added.
When option defaultroute is set to 1 the connection to the ocserv is lost
```
Got CONNECT response: HTTP/1.1 200 CONNECTED
CSTP connected. DPD 29, Keepalive 32400
DTLS handshake failed: Resource temporarily unavailable, try again.
```
Here's the routes when I use the original script (with our without $INTERNAL_IP4_ADDRESS)
There's no route on the wan link to reach the ocserv so the connection fail just after beeing established.

```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         10.x.x.x  0.0.0.0         UG    0      0        0 vpn-oc
10.x.x.x      *               255.255.0.0     U     0      0        0 vpn-oc
192.168.x.x     *               255.255.255.0   U     0      0        0 eth0.2
192.168.x.x   *               255.255.255.255 UH    0      0        0 eth0.2
192.168.x.x     *               255.255.255.0   U     0      0        0 br-lan
```
I've attached the modified script that I've used to make the default route through the openconnect link 

[Uploading vpnc-script.txt…]()

nmav, 2016-04-29:
I'm not sure "1" is valid value. According to proto_add_ipv4_route function [0] this is a mask, and it also accepts the gateway. I would expect appending the gw name to correctly add the route.

Most likely by seeing dhcp.script we need to add a route for the vpn gateway address, like:
```
proto_add_ipv4_route "0.0.0.0" 0 $INTERNAL_IP4_ADDRESS
network_get_device if_wan wan
proto_add_ipv4_route $VPNGATEWAY 32 $if_wan
```

Now after this your routing table should have a separate rule for the IP of your VPN. Is that the case?

[0]. https://github.com/stephank/luci2-netifd/blob/master/dummy/netifd-proto.sh#L120

fatchoy, 2016-04-29:
@nmav I'm sorry for the confusion, when I talk about the option defaultroute I refer to the network configuration of the openconnect interface in /etc/config/network

I will try the modification you mention tonight and report back.

The last version of the netifd-proto.sh can be found here : http://git.openwrt.org/?p=project/netifd.git;a=blob_plain;f=scripts/netifd-proto.sh;hb=HEAD
it seems there's now the parameters for the source and metric.

fatchoy, 2016-04-30:
Hello, I've modified the vpnc-script as you proposed but it doesn't add the route to the VPN Gateway, I've tried to add the gateway as parameter as follow but still not working 
```
. /lib/functions/network.sh                                             
network_get_device if_wan wan                              
network_get_gateway gw_wan wan                                          
proto_add_ipv4_route $VPNGATEWAY 32 $gw_wan $if_wan
```

nmav, 2016-05-02:
What is not working? The direct route is not added? I'll try to check it out myself but it will not be soon. If you could figure a solution for that I'd really appreciate it.

fatchoy, 2016-05-03:
I have 3 problems with the routes
1. When option defaultroute is set to 0 in /etc/config/network the default route (0.0.0.0/0) is not added even if pushed by ocserv.
The only way to set the VPN as the default route is to put defaultroute to 1 in /etc/config/network

The 2 other problems are the result of setting the VPN as the default gateway.
2. When the default route is set to the VPN, the connection is broken immediately after connection because there's no direct route to the VPN gateway through the WAN interface.
3. When disconnecting OpenConnect, the system is left without default route (problem already raise here  https://github.com/openwrt/packages/issues/1058)

To solve issue 2 and 3 I've made the following change to the vpnc-script 
Added the following at the beginning of the script

```
DEFAULT_ROUTE_FILE=/var/etc/openconnect-defaultroute

get_default_gw() {
    netstat -r -n | awk '/:/ { next; } /^(default|0\.0\.0\.0)/ { print $2; }'
}

set_vpngateway_route() {
    route add -host "$VPNGATEWAY" gw "`get_default_gw`"
    echo "`get_default_gw`" > "$DEFAULT_ROUTE_FILE"
}

del_vpngateway_route() {
    route del -host "$VPNGATEWAY"
    if [ -s "$DEFAULT_ROUTE_FILE" ]; then
        route add default gw `cat "$DEFAULT_ROUTE_FILE"`
        rm -f -- "$DEFAULT_ROUTE_FILE"
    fi
}
```
added the line `set_vpngateway_route` in the do_connect part
added `del_vpngateway_route` in the do_disconnect

nmav, 2016-05-14:
I'd prefer to use the netifd commands to setup that, but I have no idea how to do that. I've asked in the developers mailing list.

closed 2016-05-22 by nmav

gnutlsmirror, 2016-05-22:
This was addressed by the patch above. @fatchoy I've verified it locally, but I'd appreciate if you could confirm that it addresses your issue.

nmav, 2016-05-22:
This was addressed by the patch above. @fatchoy I've verified it locally, but I'd appreciate if you could confirm that it addresses your issue.

fatchoy, 2016-05-23:
@nmav Thanks for your update, I confirm that with the change above, the route to the VPN gateway is added correctly.
The remaining issue is when option defaultroute is set to 0 in /etc/config/network the default route (0.0.0.0/0) is not added even if pushed by ocserv.
The only way to set the VPN as the default route is to put defaultroute to 1 in /etc/config/network\
Should I open another ticket for this one?

jow-, 2016-05-23:
The very purpose of `option defaultroute 0` is to inhibit any default route the protocol intends to install. If you *want* a defualtroute then just do not set this option or set it to `1`.

nmav, 2016-05-23:
This option is a bit hidden but also available in the luci interface. You need to go to advanced settings.

nmav, 2016-05-23:
Thank you for testing

commit 4083de9d (2016-05-22): openconnect: use proto_add_host_dependency

This ensures that a direct route to the connected host is added
by netifd.

Resolves #2548
commit 3164b99f (2016-05-22): openconnect: use proto_add_host_dependency

This ensures that a direct route to the connected host is added
by netifd.

Resolves #2548