openwrt/packages #2551: Collectd and Luci-app-statistics segfault

TimelessNL, 2016-03-24:
Hi,

I'm running a x64 SMP target (PC Engines APU1 and APU2) and I have 3 problems:

1. collectd-mod-sensors is not visible as plugin in LuCi (cannot be enabled)
2. collectd-mod-interfaces generates wrong date (also incomplete)
3. because of some segfault the collectd crashes on every boot an needs a manual restart.
Example:
`rrdtool plugin: RRASi[  531.663917] collectd[2038]: segfault at 3130 ip 00007f17e41fb941 sp 00007f17e1b02db8 error 4 in libuClibc-0.9.33.2.so[7f17e41cf000+50000]`

Notes on 1. I've this package installed but when I go to statistics -> Setup | there is no sensors plugin to enable.

Notes on 2. The data is incomplete (no RX bytes/s) and huge numbers are displayed at the packages/s graph (ex. 2223222233222.1) 
[image]

I've reproduced this problem om 3 x64 targets. and also tried the x86 build and noticed that the collectd-mod-interfaces is working correctly here but the collectd-mod-sensors isn't listed either :(

How to reproduce:

1. Use Chaos Calmer 15.05.1 x64 generic image
2. install: 
luci-app-statistics 	        git-16.018.33482-3201903-1
collectd 	                        5.4.2-2
collectd-mod-conntrack 	5.4.2-2
collectd-mod-cpu 	        5.4.2-2
collectd-mod-interface 	5.4.2-2
collectd-mod-iwinfo     	5.4.2-2
collectd-mod-load 	        5.4.2-2
collectd-mod-memory 	5.4.2-2
collectd-mod-network 	5.4.2-2
collectd-mod-rrdtool 	5.4.2-2
collectd-mod-syslog 	5.4.2-2
collectd-mod-uptime 	5.4.2-2
collectd-mod-sensors 	5.4.2-2
3. enable all the visible plugins (except sensors which is not visible)
4. restart target
5. after target restart, restart collectd service because it crashed
6. see the outcome in the graphs

I hope my information is understandable, if something is missing please let me know. 

Regards

TimelessNL, 2016-03-24:
I tried DESIGNATED DRIVER (Bleeding Edge, r49066) x64 SMP and noticed that everything seems to be working in trunk. collectd-mod-interfaces is reporting expected results and collect-mod-sensors is also listed in LuCi and works.

would it be possible to backport Collectd to CC?

unrealwzg, 2016-06-11:
same issue here, can anyone follow this issue?

dl12345, 2016-11-04:
The trunk version of collectd compiles and runs fine on CC....no porting per se required. Just overwrite with the trunk version, compile and upgrade.

TimelessNL, 2016-11-28:
Mmh. then this issue can be closed I think.

closed 2016-11-28 by hnyman