openwrt/packages #2559: squid on 15.05.1 & imagebuilder problem

reinerotto, 2016-03-26:
When using imagebuilder on 15.05.1 follwoing problem with squid:

make image PROFILE=MYNETN600 PACKAGES="squid"
.....
Collected errors:
 * opkg_install_cmd: Cannot install package squid.
make[2]: *** [package_install] Error 255
make[2]: Leaving directory `/etc/openwrt/OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64'
make[1]: *** [_call_image] Error 2
make[1]: Leaving directory `/etc/openwrt/OpenWrt-ImageBuilder-15.05.1-ar71xx-generic.Linux-x86_64'
make: *** [image] Error 2

Worked on 15.05.

diizzyy, 2017-06-09:
@hnyman @thess 
15.05.1 is unsupported, please close.

@reinerotto 
Please try this on LEDE and if you're still experiencing issue create a new issue.

diogosimao, 2018-03-22:
Hi @reinerotto 
Would you confirm that squid should work properly on LEDE if I follow [instructions](https://openwrt.org/docs/guide-user/services/proxy/proxy.squid) for transparent mode on devices with enough flash space?

I am experience major router slow down and pages won't load after starting squid e apply firewall rule.

reinerotto, 2018-03-26:
I have squid-intercept running on several different router families, >= 128MB RAM, >=8MB flash.
Both as proxy-only with small RAM, or proxy-cache using SSD.
However, I tailored the Make file to drop unnecessary modules, i.e. most of the auths etc.
And I do _NOT_ use the clumsy setup you referenced, but the standard LINUX-like config/setup, using
/etc/squid.conf only
squid is a memory hog, so you have to make some efforts to tailor it to your needs.

closed 2018-08-17 by dibdot