openwrt/packages #2562: dropbear listens on public ipv6 connection after process restart

Aditza2015, 2016-03-27:
What do i have to do so that dropbear listens ONLY on ipv4 connections?

in the allowed parameters description i did not find this
https://wiki.openwrt.org/doc/uci/dropbear

when my router starts up, dropbear runs like this
` 4732 root      1220 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 192.168.0.1:22 -p fd41:_ETC_ipv6_:22 -K 300
`
and after the IPv6 PPPoE connection starts up, if i run the command

/etc/init.d/dropbear restart

then dropbear starts to listen on the public-accessible IPv6 address
 `7598 root      1152 S    /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 192.168.0.1:22 -p 2a02:2f0d:_ETC_ipv6_:22 -p fd41:_ETC_ipv6_:22 -K 300
`
netstat info:
```
tcp        0      0 192.168.0.1:22          0.0.0.0:*               LISTEN      7598/dropbear
tcp        0      0 fd41:_ETC_ipv6_:22    :::*                    LISTEN      7598/dropbear
tcp        0      0 2a02:2f0d:_ETC_ipv6_:22 :::*                    LISTEN      7598/dropbear
```

since i did not find how to prevent dropbear from listening on the public IPv6 address, i created firewall rules to block anything incoming to destination port 22, both on INPUT and on the FORWARD chains

block_incoming_ssh - Any TCP, UDP - From any host in wan To any router IP at port 22 on this device
Discard input

block_incoming_ssh_fw - Any TCP, UDP - From any host in wan To any host, port 22 in any zone
Discard forward

root@MyRouter-v2:/etc/config# iptables-save | grep 22
[...]
```
-A zone_wan_forward -p tcp -m tcp --dport 22 -m comment --comment block_incoming_ssh_fw -j DROP
-A zone_wan_forward -p udp -m udp --dport 22 -m comment --comment block_incoming_ssh_fw -j DROP
-A zone_wan_input -p tcp -m tcp --dport 22 -m comment --comment block_incoming_ssh -j DROP
-A zone_wan_input -p udp -m udp --dport 22 -m comment --comment block_incoming_ssh -j DROP
```
root@MyRouter-v2:/etc/config# ip6tables-save | grep 22
[...]
```
-A zone_wan_forward -p tcp -m tcp --dport 22 -m comment --comment block_incoming_ssh_fw -j DROP
-A zone_wan_forward -p udp -m udp --dport 22 -m comment --comment block_incoming_ssh_fw -j DROP
-A zone_wan_input -p tcp -m tcp --dport 22 -m comment --comment block_incoming_ssh -j DROP
-A zone_wan_input -p udp -m udp --dport 22 -m comment --comment block_incoming_ssh -j DROP
```

my network and dropbear configuration:

/etc/dropbear/config 
```
config dropbear
        option PasswordAuth 'on'
        option Port '22'
        option Interface 'lan'

```
and relevant bits of  /etc/config/network
```
config interface 'lan'
        option ifname 'eth1'
        option force_link '1'
        option type 'bridge'
        option proto 'static'
        option netmask '255.255.255.0'
        option stp '1'
        option ipaddr '192.168.0.1'
        option ip6assign '64'

config interface 'wan'
        option ifname 'eth0'
        option _orig_ifname 'eth0'
        option _orig_bridge 'false'
        option proto 'pppoe'
        option username '_my_pppoe_username_here'
        option password '__my_pppoe_pw_here'
        option peerdns '0'
        option dns '_my_ipv4_dns_servers_here _my_ipv6_dns_servers_here'
        option ipv6 'auto'
        option reqprefix 'auto'
        option reqaddress 'force'
```

Aditza2015, 2016-03-27:
P.S. ifconfig info:
NOTE: dropbear is listening on the IPv6 address from br-lan... but that one is still public-accessible

root@MyRouter-v2:~# ifconfig
br-lan    Link encap:Ethernet  HWaddr xxxxxxxxxxxxxxxxxxxxx
          inet addr:192.168.0.1  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::_ETC_ipv6_/64 Scope:Link
          inet6 addr: fd41:_ETC_ipv6_::1/64 Scope:Global
          inet6 addr: 2a02:2f0d:_ETC_ipv6_/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:234667 errors:0 dropped:0 overruns:0 frame:0
          TX packets:153136 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0
          RX bytes:287862887 (274.5 MiB)  TX bytes:26678071 (25.4 MiB)

pppoe-wan Link encap:Point-to-Point Protocol
          inet addr:xxxxxxxxxxxxxxx  P-t-P:10.0.0.1  Mask:255.255.255.255
          inet6 addr: fe80::_ETC_ipv6_/10 Scope:Link
          inet6 addr: 2a02:2f0d:_ETC_ipv6_/128 Scope:Global
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:1492  Metric:1
          RX packets:146565 errors:0 dropped:0 overruns:0 frame:0
          TX packets:223867 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:3
          RX bytes:20771573 (19.8 MiB)  TX bytes:286770435 (273.4 MiB)

stintel, 2016-03-28:
Dropbear is part of OpenWrt base, not the packages feed. Please report on Trac and close this issue.

yousong, 2016-03-28:
Hi, @Aditza2015 , dropbear is maintained as in the `base` packages set and is in the OpenWrt main repo.  Opening ticket at http://dev.openwrt.org is expected for such packages.

As for the dropbear issue, the default firewall policy of `zone_wan_input` chain should already be `REJECT`, and this can be confirmed here with `ip6tables -L zone_wan_input -nv`

I will close this ticket.  Please open a ticket at dev.openwrt.org when you feel the need.

closed 2016-03-28 by yousong