openwrt/packages #2564: building image with dnscrypt failed

Althathwe, 2016-03-30:
```
make[3]: Entering directory '/home/user/openwrt/feeds/packages/net/dnscrypt-proxy'
mkdir -p /home/user/openwrt/dl
/home/user/openwrt/scripts/download.pl "/home/user/openwrt/dl" "dnscrypt-proxy-1.4.3.tar.gz" "54d172236a8f321fb5689ff81767f1ba" "" "http://download.dnscrypt.org/dnscrypt-proxy"
URL transformed to HTTPS due to an HSTS policy
--2016-03-30 16:51:09--  https://download.dnscrypt.org/dnscrypt-proxy/dnscrypt-proxy-1.4.3.tar.gz
Resolving download.dnscrypt.org (download.dnscrypt.org)... 91.121.49.42
Connecting to download.dnscrypt.org (download.dnscrypt.org)|91.121.49.42|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://dnscrypt.org/ [following]
--2016-03-30 16:51:09--  https://dnscrypt.org/
Resolving dnscrypt.org (dnscrypt.org)... 91.121.49.42
Connecting to dnscrypt.org (dnscrypt.org)|91.121.49.42|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 26593 (26K) [text/html]
Saving to: 'STDOUT'

-                   100%[===================>]  25.97K  --.-KB/s    in 0.09s   

2016-03-30 16:51:10 (297 KB/s) - written to stdout [26593/26593]

MD5 sum of the downloaded file does not match (file: 7b920dd6b67aebc0803591b8513120ea, requested: 54d172236a8f321fb5689ff81767f1ba) - deleting download.
--2016-03-30 16:51:10--  http://mirror2.openwrt.org/sources/dnscrypt-proxy-1.4.3.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-03-30 16:51:10 ERROR 404: Not Found.

Download failed.
--2016-03-30 16:51:10--  http://downloads.openwrt.org/sources/dnscrypt-proxy-1.4.3.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2016-03-30 16:51:10 ERROR 404: Not Found.

Download failed.
No more mirrors to try - giving up.
Makefile:118: recipe for target '/home/user/openwrt/dl/dnscrypt-proxy-1.4.3.tar.gz' failed
make[3]: *** [/home/user/openwrt/dl/dnscrypt-proxy-1.4.3.tar.gz] Error 2
make[3]: Leaving directory '/home/user/openwrt/feeds/packages/net/dnscrypt-proxy'
package/Makefile:191: recipe for target 'package/feeds/packages/dnscrypt-proxy/compile' failed
make[2]: *** [package/feeds/packages/dnscrypt-proxy/compile] Error 2
make[2]: Leaving directory '/home/user/openwrt'
package/Makefile:188: recipe for target '/home/user/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile' failed
make[1]: *** [/home/user/openwrt/staging_dir/target-mipsel_24kec+dsp_uClibc-0.9.33.2/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/openwrt'
/home/user/openwrt/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

Reason: https://download.dnscrypt.org/dnscrypt-proxy/dnscrypt-proxy-1.4.3.tar.gz was moved to https://download.dnscrypt.org/dnscrypt-proxy/old/dnscrypt-proxy-1.4.3.tar.gz. Because its old.

hnyman, 2016-03-30:
cc to maintainer: @damianorenfer

yousong, 2016-08-11:
Close this as the dnscrypt-proxy has been bumped to version 1.6.1 (it's 1.4.3 in this issue).

closed 2016-08-11 by yousong