openwrt/packages #2579: Adblock addtition rlues subscription,could i sucess

likemelikeit, 2016-04-06:


hnyman, 2016-04-06:
I have no idea what you do mean by that...
Please clarify the title and explain your question more clearly

likemelikeit, 2016-04-07:
@hnyman I'm sorry for my poor expression，the trouble is： I want to use the adblock rule (https://raw.githubusercontent.com/cjx82630/cjxlist/master/cjxlist.txt) in adblock package, the key point is it do not support by default.
I have try to add the bellow to `adblock.conf` before building package ,but it didn't work.
 `config source 'CJX'
	option enabled '1'
	option adb_src 'https://raw.githubusercontent.com/cjx82630/cjxlist/master/cjxlist.txt&ruleset=rset_CJX'
	option adb_srcdesc 'personal use'
`

hnyman, 2016-04-07:
I think that you also need to add the correct awk string definition for parsing the list contents (as various blocklists have different formats). See here:
https://github.com/openwrt/packages/blob/master/net/adblock/files/adblock-helper.sh#L170

@dibdot should know better.

dibdot, 2016-04-07:
@hnyman @likemelikeit In adblock 1.0.x this is (partly) hardcoded in adblock-helper.sh, you cannot add new source sites on your own. Please use the static blacklist for manual domain additions.
In the next release cycle adblock 1.1.x this will be more flexible and the user can add new sources as you've mentioned above.
Regarding your posted list: It's a stripped down easylist. Adblock already includes easylist/ruadlist, see here: [https://easylist-downloads.adblockplus.org/ruadlist+easylist.txt](url).
As adblock works on domain level, only "full domains" can be taken from the easylist. Full domains in easylist are qualified by "|| <domain>^" ... in your list there is not a single "full domain".

likemelikeit, 2016-04-07:
@hnyman @dibdot Thanks very much.

closed 2016-04-07 by likemelikeit