openwrt/packages #2582: [Orig: #21654] Aiccu should depend on ip or ip-full

leleobhz, 2016-04-08:
I'm affected by the bug https://dev.openwrt.org/ticket/21654 and i reproduced it:

```
root@amora:~# opkg install ip-full
Installing ip-full (4.0.0-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/brcm2708/bcm2708/packages/base/ip-full_4.0.0-1_brcm2708.ipk.
Collected errors:
 * check_data_file_clashes: Package ip-full wants to install file /usr/sbin/ip
	But that file is already provided by package  * ip
 * opkg_install_cmd: Cannot install package ip-full.
root@amora:~# opkg remove ip
No packages removed.
Collected errors:
 * print_dependents_warning: Package ip is depended upon by packages:
 * print_dependents_warning: 	aiccu
 * print_dependents_warning: These might cease to work if package ip is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.
root@amora:~# opkg --force-removal-of-dependent-packages remove ip
Removing package aiccu from root...
Removing package ip from root...
Not deleting modified conffile /etc/iproute2/rt_tables.
root@amora:~# opkg install ip-full
Installing ip-full (4.0.0-1) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/brcm2708/bcm2708/packages/base/ip-full_4.0.0-1_brcm2708.ipk.
Configuring ip-full.
root@amora:~# opkg install aiccu
Installing aiccu (20070115-12) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/brcm2708/bcm2708/packages/packages/aiccu_20070115-12_brcm2708.ipk.
Installing ip (4.0.0-1) to root...
Collected errors:
 * check_conflicts_for: The following packages conflict with ip:
 * check_conflicts_for: 	ip-full * 
 * opkg_install_cmd: Cannot install package aiccu.
```

It's possible to add ip and ip-full to satisfied conditions?

luizluca, 2016-04-12:
I guess it is more an ip-full issue as it does not "provides ip". If so, aiccu could simply continue to require ip.

leleobhz, 2016-04-27:
@luizluca Indeed. But ip-full reports are covered by  this bugtrack?

leleobhz, 2016-06-02:
root@amora:~# opkg install tayga
Installing tayga (0.9.2-2) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05/brcm2708/bcm2708/packages/packages/tayga_0.9.2-2_brcm2708.ipk.
Installing ip (4.0.0-1) to root...
Collected errors:
 * check_conflicts_for: The following packages conflict with ip:
 * check_conflicts_for: 	ip-full * 
 * opkg_install_cmd: Cannot install package tayga.

Again, ip-full superseeds ip package. Its not possible to fill this dependency?

yousong, 2016-08-11:
ping aiccu and tayga's current maintainer @oskar456 ;)

oskar456, 2016-08-11:
I've found probably same issue with core package `openvpn` which is probably the only core package depending (at least conditionally) on `ip`. Other packages depending on `+ip` are:
`mwan3`, `opennhrp`, `strongswan`, `shadowsocks-libev`, `chaosvpn`

Is there any way how to depend on either `ip` or `ip-full`?
I have no idea.

oskar456, 2016-08-11:
It could probably work like this:

    DEPENDS:=… @(PACKAGE_ip||PACKAGE_ip-full)

But it would mean the package is not visible unless `ip` or `ip-full` is selected.

stintel, 2016-10-31:
Once the package PROVIDES handling has been improved, this can be properly fixed. See http://lists.infradead.org/pipermail/lede-dev/2016-October/003643.html

diizzyy, 2017-06-09:
@oskar456 @leleobhz 
Is still package still broken?

oskar456, 2017-06-09:
AFAIK yes. And since the SixXS service has been shut down on 2017-06-06 and the server implementation of TIC/AYIYA protocol is SixXS proprietary, I don't see a point in trying to fix it.

diizzyy, 2017-06-09:
@oskar456 
I see, please close this and provide a pull request removing aiccu.

oskar456, 2017-06-09:
Good idea, pull request submitted. So long and thanks for all the tunnels.

diizzyy, 2017-06-09:
@thess @hnyman 
Please close this.

PR #4464 (2017-06-12): aiccu: remove package
Maintainer: Ondřej Caletka / @oskar456
Compile tested: N/A
Run tested: N/A

Description:
Aiccu is a client utility used to connect to the SixXS platform. As the SixXS platform has been [sunset on 2017-06-06][1] and the server parts of TIC/AYIYA protocol implementations haven't been published, there's no point to maintain aiccu package anymore.

[1]: https://www.sixxs.net/sunset/

Closes #2582
closed 2017-06-09 by hnyman