openwrt/packages #2585: Where is the official xorg feed?

mkschreder, 2016-04-09:
The feed in feeds.conf.default appears to be broken (svn://svn.openwrt.org/openwrt/feeds/xorg). 

Is there an up-to-date xorg feed somewhere?

hnyman, 2016-04-10:
To my knowledge the xorg feed is unmaintained, last update was 4 years ago. I doubt if there is much anything usable for CC15.05.1 or trunk. The feed was last updated when Backfire 10.03.1 was the current release, so really ancient versions there.

It can still be browsed from the dev.openwrt.org site:
https://dev.openwrt.org/log/feeds/xorg
https://dev.openwrt.org/browser/feeds/xorg

The commented/disabled feed definition should probably be removed from feeds.conf.default

titobrasolin, 2016-04-14:
The following alternative feed definition still works:
```
src-svn xorg svn://svn.mein.io/openwrt/feeds/xorg
```
See also:
https://www.mail-archive.com/openwrt-devel%40lists.openwrt.org/msg38483.html

mkschreder, 2016-04-27:
Great. I have pushed to git also: https://github.com/mkschreder/openwrt-xorg-feed

Thanks for posting this here!

titobrasolin, 2016-04-29:
@mkschreder you are welcome! By the way, do you accept pull requests? I worked on a few patches (mostly related to Qt4) that I would like to submit for review and inclusion.

mkschreder, 2016-04-29:
If you want to post pull requests into my copy then you're welcome. I can
host the feed on my account for now.

closed 2016-08-11 by yousong

mirko, 2016-09-08:
I maintained the xorg feed and currently maintain the official video feed which qt4 is now also part of.
I'm happy to accept pull requests / patches.

mkschreder, 2016-09-08:
Can we merge our feeds? You are welcome to take packages from this repo and
maintain them as well: https://github.com/mkschreder/openwrt-xorg-feed ..

mirko, 2016-09-08:
I wouldn't know what to take and what not to. The xorg stuff is outdated and unmaintained, I'm currently working on getting a minimal recent xorg-server ported.
The rest - as in: maintainted and used stuff besides xorg - I'm happy to take over. Pull requests preferred.