openwrt/packages #2597: Please update crelay package to new upstream version

ondrej1024, 2016-04-14:
@tess: referring to #1662, there is a new release of the crelay control daemon. 
https://github.com/ondrej1024/crelay/releases/tag/V0.10

Please update the related OpenWRT package.
Thanks

hnyman, 2016-04-14:
I guess you meant @thess ...

ondrej1024, 2016-04-14:
Yes, that's right. Sorry.

thess, 2016-04-14:
Got it - looking to do it soon. @ondrej1024 Anything I ought to be looking for other than simple source update. Features/modules etc ?

ondrej1024, 2016-04-15:
The new version adds support for another relay card (Sainsmart 16 channel control module) which is controlled via the USB HID API. We have already included the needed library in an earlier version, so there are no new dependencies. 
Another small change is that crelay is daemonised in daemon mode instead of running in foreground. Don't know if this needs some changes in the package.

champtar, 2016-04-15:
Hi @ondrej1024,
With procd (openwrt init) it's prefered to have the program running in the foreground

thess, 2016-04-15:
@ondrej1024 - as Etienne said, we want daemons executing in the foreground under procd. To that end, I will patch out the daemon() call in the package for now, and... I will update the package later if you would be so kind as to supply an option to run in daemon mode but stay in foreground.

thess, 2016-04-17:
Done. See https://github.com/openwrt/packages/commit/625a7e7563f23fbb97186fa059bab67de4847cdb

daemon() call patched out for now.

closed 2016-04-17 by thess

ondrej1024, 2016-04-18:
Thanks @thess. 
The daemon() function was added by a patch I accepted for crelay and I thought it would be a good idea to send the program to background when running in daemon mode. If this is not always desirable I can add a `-D` option (in addition to the existing `-d`) which starts crelay in daemon mode but remains in foreground. I will do that for the next update.