openwrt/packages #2603: latest changes/upgrade to python2.7 - python build failed (host and program to system) in ubuntu 16.04 - RC

build000, 2016-04-14:
`(...)
make[3]: Entering directory '/home/slawek/build_openwrt/openwrt-witi-trunk/feeds/packages/lang/python'
rm -rf /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python
mkdir -p /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/host /home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/packages /home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/host/packages
install -d -m0755 /home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/mk/
install -d -m0755 /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/include/ /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/lib/ /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/lib/pkgconfig
install -d -m0755 /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/lib/python2.7/
install -m0644 ./files/python-package.mk ./files/python-host.mk /home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/mk/
cp -fpR /home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.14/Python-2.7.11/ipkg-install/usr/include/python2.7 /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/include/
cp -fpR /home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/host/lib/python2.7 /home/slawek/build_openwrt/openwrt-witi-trunk/build_dir/target-mipsel_1004kc+dsp_musl-1.1.14/Python-2.7.11/ipkg-install/usr/lib/libpython2.7.so* /home/slawek/build_openwrt/openwrt-witi-trunk/tmp/stage-python/usr/lib/
cp: cannot stat '/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/host/lib/python2.7': No such file or directory
Makefile:238: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/stamp/.python_installed' failed
make[3]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/stamp/.python_installed] Error 1
make[3]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk/feeds/packages/lang/python'
package/Makefile:195: recipe for target 'package/feeds/packages/python/compile' failed
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
package/Makefile:192: recipe for target '/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/stamp/.package_compile' failed
make[1]: *** [/home/slawek/build_openwrt/openwrt-witi-trunk/staging_dir/target-mipsel_1004kc+dsp_musl-1.1.14/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/slawek/build_openwrt/openwrt-witi-trunk'
/home/slawek/build_openwrt/openwrt-witi-trunk/include/toplevel.mk:192: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
`

hnyman, 2016-04-14:
@build000 
Which exact git commit of the packages feed you are building from?

@commodo

build000, 2016-04-14:
...latest changes (python, python3, etc):
https://github.com/openwrt/packages/commit/6255aa959dd6e22854dca5f4129898d7896db890
https://github.com/openwrt/packages/commit/7289a98f4b2319c8638323b298240fce894027b1
https://github.com/openwrt/packages/commit/c8e8aae57916703b208a51d72a8259232f9b6693
https://github.com/openwrt/packages/commit/32db03968fdd814328156216742bc175db1237df
https://github.com/openwrt/packages/commit/32db03968fdd814328156216742bc175db1237df

commodo, 2016-04-14:
checking now ;
i'll re-spin a trunk build

i validated those changes against a 15.05 branch

build000, 2016-04-14:
of course my compilation is "trunk"
Again:
1. make dirclean
2. git pull (no any changes)
3. make package/symlinks V=s (no any changes)
4. date && make -j8 && date (its the same moment build error)
5. date && make -j1 V=s && date (its the same "build error" - post 1 - https://github.com/openwrt/packages/issues/2603#issue-148405408)

jow-, 2016-04-15:
I am hitting the same issue in a buildbot setup:

    make[2]: Entering directory `/build/sunxi_generic/build/feeds/packages/lang/python'
    rm -rf /build/sunxi_generic/build/tmp/stage-python
    mkdir -p /build/sunxi_generic/build/tmp/stage-python/host /build/sunxi_generic/build/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/packages /build/sunxi_generic/build/staging_dir/host/packages
    install -d -m0755 /build/sunxi_generic/build/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/mk/
    install -d -m0755 /build/sunxi_generic/build/tmp/stage-python/usr/include/ /build/sunxi_generic/build/tmp/stage-python/usr/lib/ /build/sunxi_generic/build/tmp/stage-python/usr/lib/pkgconfig
    install -d -m0755 /build/sunxi_generic/build/tmp/stage-python/usr/lib/python2.7/
    install -m0644 ./files/python-package.mk ./files/python-host.mk /build/sunxi_generic/build/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/mk/
    cp -fpR /build/sunxi_generic/build/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/Python-2.7.11/ipkg-install/usr/include/python2.7 /build/sunxi_generic/build/tmp/stage-python/usr/include/
    cp -fpR /build/sunxi_generic/build/staging_dir/host/lib/python2.7 /build/sunxi_generic/build/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/Python-2.7.11/ipkg-install/usr/lib/libpython2.7.so* /build/sunxi_generic/build/tmp/stage-python/usr/lib/
    cp: cannot stat `/build/sunxi_generic/build/staging_dir/host/lib/python2.7': No such file or directory
    make[2]: *** [/build/sunxi_generic/build/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.14_eabi/stamp/.python_installed] Error 1
    make[2]: Leaving directory `/build/sunxi_generic/build/feeds/packages/lang/python'
    make[1]: *** [package/feeds/packages/python/compile] Error 2
    make[1]: Leaving directory `/build/sunxi_generic/build'
    make: *** [package/compile] Fehler 2

In my case any python stuff is disabled in the config but the python build is implicitely triggered by setting `CONFIG_PACKAGE_kmod-openvswitch=m` in `.config`. Is it possible that the python build fails when it is implicitely invoked without any python related symbols in the config?

commodo, 2016-04-15:
it looks like in trunk, some build dir and staging dir stuff diverged from CC/15.05

Something like this:
http://git.openwrt.org/?p=openwrt.git;a=commitdiff;h=c5c292a7a8ad9d73a6764060abaf024e78951b8e

change is from January of this year;
looks like nbd has been doing some cleanup, and I just messed it up 
the changes look a bit quirky to me, but i won't complain

hmm; i guess i will roll-back some of my changes
there is no sense in working in an opposite direction

build000, 2016-04-15:
> Is it possible that the python build fails when it is implicitely invoked without any python related symbols in the config?

... exactly

commodo, 2016-04-15:
PRs:  https://github.com/openwrt/packages/pull/2606  and  https://github.com/openwrt/packages/pull/2605
Tested on trunk and CC branch.

build000, 2016-04-15:
I confirm: this "undo" (https://github.com/openwrt/packages/pull/2605 and https://github.com/openwrt/packages/pull/2606) fix the problem.
To check the deeper implications and potential problems for the test compiled all the packages from the "Languages/Python" marked explicitly by the command "make menuconfig". Full success - all available packages python and supplementary programs/libraries, and their dependencies built correctly, bearing fruit the creation of appropriate packages.
Application to close if the remaining content with such a solution.
In my humble opinion, these changes can be introduced into the main tree of the project ("packages") openwrt.

closed 2016-04-16 by build000