openwrt/packages #2604: luci_statistics not flushing it's cache on router reboot

kylejvrsa, 2016-04-15:
I have a usb flash drive connected to my router. I use this drive for storing all luci_statistics (rrd graphs), because this is a USB flash drive I want to limit the writes to it as they don't last long when writing allot, so I flush the collectd cache every 10mins, when I need to restart luci_statistics (/etc/luci_statistcis restart) it flushes the cache, however when I reboot the router it doesn't flush the cache it just reboots and I lose all the gathered data in the 10min.

Router: WRT1900AC
Firmware: OpenWRT 15.05

dfc-d, 2016-04-15:
Silly question, but why do you think this matters?  Do you have some
data that wear levelling doesn't work even remotely close to theory so
that even with 10 years of ridiculously frequent writes on even 8 MB
flash you wouldn't be fine?

Regards,

Daniel

kylejvrsa, 2016-04-15:
I just thought I'd mention this, as I was writing to the drive very frequently and the flash drive failed within 4 months so I increased the cache flush in the hope that writing to it may increase it's lifespan. Anyway so what is your take on the cache being flushed on reboot?

Cheers!

dfc-d, 2016-04-15:
What sort of failure for the flash drive?  i.e. I've had usb flash
drives not being written often fail due to crappy usb electronics that
do nasty things to the electronics of the drive when the device had
constant moderate cpu and higher temperatures than stock units normally
operate with.

That is I'm not convinced reduced flush is actually doing to do anything
for you.

Regards,

Daniel

kylejvrsa, 2016-04-15:
Whether I will increase the lifespan of the flash drive with increased cache flush time might help or it might not. The reason I created the issue was to find out whether the luci_statistics should flush it's cache or not. So what is your take on cache being flushed on router reboot? I would think that, that should be done even if it's just 30 seconds of cache.

dfc-d, 2016-04-15:
Sorry just noticed what I suspect is an urban legend about flash
lifespan (I've had conversations with a product manager who was
convinced that the fact that even (even years ago) cheap flash has wear
levelling built in, and I read the specs and did some calculations and
came to conclusion that unless manufacturers are managing to pass off
total fiction and not getting sued for it, that flash writes are really
a non-issue except under extremely unusual circumstances).

I can't say I've exactly researched the topic, but what little searching
I did only comes up with anecdotal theories that the cause of failure
was excessive writes.

The concept gets bandied around so much without hard data that I know
about that it smells to me like classic urban legend (albeit it's kind
of like the battery memory legend - it was once true, but not with
modern gear (although batteries are a little different as they're still
subject to maintenance issues, about which you hear a variety of stories
for what's the best way to mitigate, there seem to be a number of pet
theories about that)).

Anyway for flushing cache on reboot I don't think that's practical
unless you can arrange for the reboot to only be done by software (in
which case the shutdown of collectd should already take care of it).

If you hit the power button there isn't a means on the vast majority of
router hardware (if not all) to intercept the the button press and
perform pc-like shutdown or reboot.

Regards,

Daniel

kylejvrsa, 2016-04-15:
If a server is restarted, a service should be successfully stopped, in collectd's case I would assume it should flush the cache before it's shutdown right?

Cheers,

dfc-d, 2016-04-16:
If restarted by e.g. reboot (but not reboot -f that bypasses
initscripts), it *ought* to flush the cache because it should be
shutdown cleanly.

Regards,

Daniel

hnyman, 2016-04-16:
This might have been fixed in DD trunk by https://github.com/openwrt/packages/commit/8228cee3f34bbae17a1625fe10e9544dbef01581
(The commit added a STOP value for the collectd init script, so that a shutdown will cause the service to stop. But I haven't tested if that really works for collectd.)

kylejvrsa, 2016-04-18:
@dfc-d: I use luci to reboot or reboot from terminal.

@hnyman: Cool! I can't test it as I am running 15.05, been pushing for us to get another device here so I can test trunk. Would be really cool if it is indeed fixed as that is the expected behaviour. 

Thanks to both of you for responding :)

hnyman, 2016-04-18:
@kylejvrsa 

You can easily test it in your 15.05 router. Just check the change.... 
It is just one line added to the text file /etc/init.d/collectd
The line "STOP=10" has been added after that START line

You can edit the file by yourself
https://github.com/openwrt/packages/blob/8228cee3f34bbae17a1625fe10e9544dbef01581/utils/collectd/files/collectd.init

diizzyy, 2017-06-09:
@hnyman 
Please close this, wrong repo and 15.X is unsupported.

closed 2017-06-09 by hnyman