openwrt/packages #2618: there is no ffmpeg—custom in menuconfig

maotouying, 2016-04-21:
When I compile ffmpeg on my openwrt, after make menuconfig there is no ffmpeg-custom in Libraries（there are ffmpeg-mini，ffmpeg-full), Is there someone know what is the reason?

thess, 2016-04-21:
If you are building OpenWrt with CONFIG_ALL, there is no custom ffmpeg. Just the standard 3 configs are produced. Custom ffmpeg does not have a "default" configuration - you must tailor it for your needs independently of full builds of everything. If you have a specific configuration in mind (codecs, parsers, protocols, etc.), put together a proposal for a 4th set of config options and we will consider it if it has general use.

maotouying, 2016-04-22:
Are there a example setting for use custom ffmpeg? beacuse I want to play m4a music, but I can't do that use ffmpeg-full, I want to use custom ffmpeg and add libsdk-acc to try it.

thess, 2016-04-22:
If you are looking for help building a custom ffmpeg library, I would suggest you look to the fmpeg and/or the openwrt support forums for additional help on options.

closed 2016-04-22 by thess