openwrt/packages #2624: Tor Hidden service hostname resets after reboot

Smim0, 2016-04-25:
Hello,

I installed tor on my router Huawei EchoLife HG556a Ver A with the lastest openwrt version Chaos Calmer 10.05.1 and I activated a hidden service to ssh into the router from outside my network. Tor works fine but it generates a new hostname/private_key after every reboot, I looked all around and it seems that nobody has this issue. Can you please help?

Smim0, 2016-04-25:
by the way, /etc/init.d/tor restart does not reset the hostname. It only does if i reboot the router

hauke, 2017-01-14:
You should store the key on some storage which gets preserved over reboots

EricLuehrsen, 2017-06-01:
Probably these two are related: https://github.com/openwrt/packages/issues/4391

rfc2822, 2017-06-02:
I have moved the hidden service directory to my mSATA SSD and now it works :)

Jookia, 2017-07-22:
I think it should be default to have it stored on the overlay. I don't think it's expected behaviour for hidden service keys to reset on reboot- it defeats the purpose of having a hidden service.

nwf, 2017-07-27:
There should be no concern of flash durability storing the hidserv director(y|ies) in the overlay; AFAIK the private key and corresponding hostname are written at initialization and, thereafter, the directories are read-only.

neheb, 2020-01-12:
Is this still an issue?

Jookia, 2020-01-12:
I haven't used Tor on OpenWrt for a long time, but unless the package has specific support for this then I would think not

Smim0, 2020-01-12:
The issue is solved

closed 2020-01-12 by neheb