openwrt/packages #2630: flac: build fails if CONFIG_DEBUG is set

dangowrt, 2016-04-27:
If `CONFIG_DEBUG=y` in OpenWrt's `.config` building the flac packages fails:
```
make[3]: Entering directory '/usr/src/openwrt/feeds/packages/libs/flac'
CFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1:flac-1.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -I/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/usr/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/usr/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/include/fortify -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/include " CXXFLAGS="-Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1:flac-1.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -I/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/usr/include -I/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/usr/include -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/include/fortify -I/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/include " LDFLAGS="-L/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/usr/lib -L/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/lib -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/usr/lib -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-5.3.0_musl-1.1.14/lib -znow -zrelro " make -j1 -C /usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/. AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1:flac-1.3.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CROSS="x86_64-openwrt-linux-musl-" ARCH="x86_64" ;
make[4]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1'
make  all-recursive
make[5]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1'
Making all in include
make[6]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include'
Making all in FLAC
make[7]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/FLAC'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/FLAC'
Making all in share
make[7]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share'
Making all in grabbag
make[8]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share/grabbag'
make[8]: Nothing to be done for 'all'.
make[8]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share/grabbag'
make[8]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share'
make[8]: Nothing to be done for 'all-am'.
make[8]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share'
make[7]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/share'
Making all in test_libs_common
make[7]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/test_libs_common'
make[7]: Nothing to be done for 'all'.
make[7]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include/test_libs_common'
make[7]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include'
make[6]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/include'
Making all in m4
make[6]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/m4'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/m4'
Making all in src
make[6]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/src'
Making all in share
make[7]: Entering directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/src/share'
  CC       getopt/getopt.lo
x86_64-openwrt-linux-musl-gcc: error: 3: No such file or directory
Makefile:633: recipe for target 'getopt/getopt.lo' failed
make[7]: *** [getopt/getopt.lo] Error 1
make[7]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/src/share'
Makefile:416: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/src'
Makefile:481: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1'
Makefile:413: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1'
Makefile:75: recipe for target '/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/.built' failed
make[3]: *** [/usr/src/openwrt/build_dir/target-x86_64_musl-1.1.14/flac-1.3.1/.built] Error 2
make[3]: Leaving directory '/usr/src/openwrt/feeds/packages/libs/flac'
package/Makefile:196: recipe for target 'package/feeds/packages/flac/compile' failed
make[2]: *** [package/feeds/packages/flac/compile] Error 2
make[2]: Leaving directory '/usr/src/openwrt'
package/Makefile:193: recipe for target '/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/stamp/.package_compile' failed
make[1]: *** [/usr/src/openwrt/staging_dir/target-x86_64_musl-1.1.14/stamp/.package_compile] Error 2
make[1]: Leaving directory '/usr/src/openwrt'
/usr/src/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
```

thess, 2016-04-27:
Amazing! Somehow the auto-majik stuff turned the -g3 (from CONFIG_DEBUG) into 3 in the CFLAGS.

... Working on it.

thess, 2016-04-27:
Fixed via 973556f29281d6ff85c0af36d3fb305992beb431

May not be entirely trim and correct, but it ameliorates the problem.

closed 2016-04-27 by thess