openwrt/packages #2633: node.js doesn't run on AR9331 : SIGILL , Illegal instruction

captcha1, 2016-04-28:
I think this is an "endian" issue.  AR9331 is big endian.  I tried on an "Linkit Smart 7688" (MediaTek) with CC 15.05.1, which is little endian, and it runs fine.

I tried in on an "Onion Omega" running CC 15.05.1 and a Carambola2 running "Bleeding Edge r49221" and both give SIGILL .

I tried to compile it myself with various endian flags, but still get SIGILL.  I'll keep trying...

jow-, 2016-04-28:
It can also be related to floating point instructions. If it would be an endian issue you wouldn't even be able to run the binary up to the point that it throws a SIGILL.

captcha1, 2016-04-30:
The Onion.io people have a nodejs build that works :

http://repo.onion.io/omega/packages/libv8_0.1-1_ar71xx.ipk

http://repo.onion.io/omega/packages/nodejs_v4.3.1-1_ar71xx.ipk

nxhack, 2016-05-02:
Is "/sys/kernel/debug/mips/fpuemustats" directory exists?
If not exist, MIPS_FPU_EMULATOR is not enabled.

captcha1, 2016-05-04:
Yes, I think that's the problem : to run node.js , I need to make a kernel
w/ CONFIG_MIPS_FPU_EMULATOR=y (this web page confirms this :
http://www.egrep.jp/wiki/index.php/OpenWrt_for_Arduino_Yun_cheat_sheet ).

captcha1, 2016-05-05:
BTW, I also filed a bug at https://dev.openwrt.org/ticket/22293 .  Dunno
which is a better place for this.

 As discussed in this web page : https://github.com/nodejs/node/issues/4818
 , I ran "make kernel_menuconfig" and selected the "Enable FPU emulation".
 Built it and node.js worked (I used the node.js v4.3.1 from Onion :
 http://repo.onion.io/omega/packages/nodejs_v4.3.1-1_ar71xx.ipk ).

 **However**, in linux-3.18.29/arch/mips/math-emu/cp1emu.c , I found this
 comment :

 Note if you know that you won't have an FPU, then you'll get much better
 performance by compiling with -msoft-float

So a better fix is to compile node.js with -msoft-float .

Though I suppose it would nice to have both -msoft-float and
MIPS_FPU_EMULATOR=y so that it runs one way or the other.

I don't know why the Linkit 7688 builds with MIPS_FPU_EMULATOR=y , but the
AR9331 does not.

Thanks,
Ken

JohannesRudolph, 2016-06-13:
Has anyone tried compiling node.js with -msoft-float instead an run on AR9331 board without MIPS_FPU_EMULATOR=y yet?

vankai, 2017-04-15:
Using [http://repo.onion.io/omega/packages/nodejs_0.1-1_ar71xx.ipk](http://repo.onion.io/omega/packages/nodejs_0.1-1_ar71xx.ipk) and [http://repo.onion.io/omega/packages/libv8_0.1-1_ar71xx.ipk](http://repo.onion.io/omega/packages/libv8_0.1-1_ar71xx.ipk) solved my problem, not V4.5.0.

diizzyy, 2017-06-02:
@captcha1 @vankai 
Is this still an issue?

NeoRaider, 2017-06-02:
NodeJS has dropped MIPS soft-float support long ago. Running a kernel with FPU emulation is the only way to use node on ar71xx.

diizzyy, 2017-06-02:
@thess @hnyman 
Can you close this as @NeoRaider has explained the issue?

closed 2017-06-02 by NeoRaider