openwrt/packages #2636: ddns-scripts: allow default cert configuration for use_https

mrjoel, 2016-04-28:
The use_https option requires the explicit setting of cacert, however using wget and ca-certificates, the certs are linked and available by default. Instead of erroring out with the error of "No valid certificate(s) found", I'd like it to fall through without any special flag, thereby using the system configured certs (at least for wget). The cacerts config value still is useful and makes sense as an override, or if needed for non-wget clients.

Note that this is different than the IGNORE, since I'd still like the cert chain to be validated, I'm just fine using the system cert store as configured.

As a "workaround" to the above issues, I'm setting the https:// explicitly in update_url, and not using the use_https option at all. In fact, I'd prefer to explicitly set it that way anyway, so it's clear to a user that https will be used. In inspection of the scripts, the 'use_https' user configuration option seems like a bit of a kludge. It appears that it is only used to test various http clients (wget, curl, busybox, etc.) and their support. It helps to provide clearer error messages, but it seems that it could be determined internally by the script (set locally based on whether $update_url contains "https://") instead of requiring separate user configuration.

chris5560, 2016-04-30:
If you look into current code line 655-665 and 693-703 in dynamic_dns_functions.sh use_https is allowed without setting cacart.

mrjoel, 2016-05-02:
Those are exactly the sections I was looking at and there are only three options that I see:
    * cacerts is "IGNORE"
    * cacerts is a file
    * cacerts is a directory

If cacerts isn't any of those, the final test checks ( elif [ -n "$cacert" ] ) causes the failure (logs at 14 which causes the exit) when cacerts is unset. In looking at it, I suppose setting cacerts to "dummy" as a non-existent file/directory may sidestep the issue. I'll try to get a patch together for clarifying what I'd expect.

The more I look at this though, the more I wonder what the purpose of a user settable use_https value is, separate from whether http or https is set in the update url string. Would you consider patches to rework that into an internal variable determined by the resulting update url scheme used?

mrjoel, 2016-05-02:
Looking at the conditions against, it looks like it's correct and should only fail if a non-file/directory value for cacert is provided, and ignore if not set... I'll take a look again at what conditions caused the issue to occur. I'd still like your thoughts on internal vs. user visible location of the setting.

yousong, 2016-08-11:
Close this as it looks like the code is "correct" for the issue originally reported and there is no update for more than 3 months

closed 2016-08-11 by yousong