openwrt/packages #2637: ddns-scripts: don't fail if domain can't be resolved

mrjoel, 2016-04-28:
The ddns script shouldn't fail if the initial nslookup fails. I'm using duckdns.org and they offer the ability to clear the IP address (through web site or via the clear=true request param). Once cleared, the domain is still linked to my account and available for setting, it just doesn't return an A/AAAA record while cleared.

The result of that is that for a new/cleared domain, I'd like the script to run and do the setting of the value, however since no records are in place, the nslookup fails. A warning would certainly make sense, however failing outright prevents the next step (actually doing the setting) which would resolve the issue.

mrjoel, 2016-04-29:
In a similar vein, when setting force_ipversion and falling back to the busybox nslookup (which doesn't support AAAA record lookup) causes a failure. I typically have the 'drill' DNS lookup utility installed, which has both "force lookups over IPv6" and record type lookups. For forced mode, it would be ideal for both to be used, but only using the AAAA record type without requiring IPv6 transit for the DNS lookup is likely safer during transition period. This allows those with non-IPv6 resolvers to still lookup the AAAA record and then submit over IPv6 transport to the DDNS provider.

chris5560, 2016-04-30:
- Most providers don't offer to create records via ddns updates. So lets support the majority and fail if domain not already registered/exists in the web.
- busybox nslookup supports AAAA records but does not support to explizit force the ip-version to use for communication. So force_ipversion when only busybox is installed makes no sense and must fail.
- from my knowledge 'drill' is not available as openwrt package, so no support.

mrjoel, 2016-05-02:
* Fair point as a default, but having a config option to ignore/skip the nslookup and blindly press ahead would be useful
* fair enough, definitely need assurance that the AAAA record was the one retrieved, although it doesn't actually matter if the IP transport for the DNS lookup is v4 or v6 - i.e., an AAAA record looked up over IPv4 should be good enough, correct?
* Drill is provided as a package within the ldns source tree, https://github.com/openwrt/packages/blob/master/libs/ldns/Makefile

yousong, 2016-08-11:
Close this since it is referred to in the commit message as resolved.

closed 2016-08-11 by yousong