openwrt/packages #2639: How to make sure maintainer sees PRs etc

dfc-d, 2016-04-29:
Hi,

Is there a good way on github to make sure a maintainer sees PRs and such (i.e. to find their github username and somehow make sure they get the notification?)

I know this isn't really an 'issue' but I don't know where to ask it, otherwise.

luizluca, 2016-04-29:
With github hooks, it could check every PR, look for Makefiles and ping maintainers, either by email or mentioning them in a comment (or both).

We could search github account by email using:

https://github.com/search?utf8=✓&q=luizluca%40gmail.com++type%3Auser&type=Users&ref=searchresults
(Or something more deterministic)

This could also work with issues, matching words in issue title with package names.

Webhooks calls a foreign site, so it will need a host site (OpenWRT.org?) for the scripts.

hnyman, 2016-04-29:
The easiest thing would be if the submitter alerts the maintainer when creating the PR. That requires some effort from the PR submitter, but no coding.

Maintainer's github account can be found pretty easily from the earlier commits to the package, then just add a line like "CC to maintainer @xxxxx" or something like that to the PR description, and the maintainer will get attached to the PR discussion.

Github account name differs from the MAINTAINER line in the package, so there is probably no easy way to programmatically find out the maintainer's current github user name.

luizluca, 2016-04-29:
How about adding a new package field that indicates the maintainer github account?

champtar, 2016-04-29:
hi guys,

I think we can start with PR/issue templates
https://github.com/blog/2111-issue-and-pull-request-templates

thess, 2016-04-29:
All good - I think some simple templates (issues & PR) would help to clean up some our push-back necessary on poorly formed and/or incomplete PR's.

Adding a new package Makefile item like PKG_MAINTAINER_NOTIFY would help out A LOT. We'd have a lot of package updates to do... The supposed map between PKG_MAINTAINER name/email to Github user name is non-existent in many cases. The submitter needs to figure it out from package history which may also not be possible depending on how the check-ins happened and by who.

hnyman, 2016-04-29:
I like the idea about a simple template, but I am much more hesitant regarding adding more metadata items to the packages.

oneru, 2016-04-29:
Templates for PRs and bugs could be quite helpful.  My approach has been to "watch" the whole repo, but that's really too many notifications, signal-to-noise and all that.  The extra metadata in the Make file would be worth the trouble if it meant an automated way to contact the maintainer.  Another option would be requesting a pair of github feature, watching a particular folder instead of the whole repo, and automatic notifications when a keyword is matched in a bug report.

yousong, 2016-08-11:
Now we have templates for pull requests and it works great.  Hi, @champtar , do you feel like adding another one for issues?

yousong, 2016-08-26:
I am closing this one now that we have templates for both issues and pull requests and maintainers are expected to be notified of these events at least once at creation time.

closed 2016-08-26 by yousong