openwrt/packages #2652: nut - conffiles incorrect

zyxmon, 2016-05-02:
Tabs before files should be removed
https://github.com/openwrt/packages/blob/master/net/nut/Makefile#L69
https://github.com/openwrt/packages/blob/master/net/nut/Makefile#L126
https://github.com/openwrt/packages/blob/master/net/nut/Makefile#L155

dfc-d, 2016-05-02:
On 16-05-02 11:25 AM, zyxmon wrote:
> Tabs before files should be removed

It works; is there any written policy or clearly defined standard in the
regard (I've seen both ways in the packages feed in other packages), or
is this just random opinion?

Regards,

Daniel

zyxmon, 2016-05-02:
Just unpack nut_*.ipk and check - there is no conffile in control archive. So it does not work.
I have checked this file - https://downloads.openwrt.org/snapshots/trunk/ramips/rt3883/packages/packages/nut_2.7.4-1_ramips_24kec.ipk
I guess - this script should be modified - https://github.com/openwrt/openwrt/blob/master/scripts/ipkg-build#L58 and leading white spaces removed from $CONTROL/conffiles
But it is much easier not to uses tabs and spaces in `/conffiles`sections!

dfc-d, 2016-05-02:
Thank you for the report; I think it has gone unnoticed because it does
not cause a build failure, but it seems the results are actually incorrect.

I'll have to find the other packages that do this too, because they
should all be updated!

Regards,

Daniel

PR #2655 (2016-05-03): net/nut: Remove whitespace before conffiles
Closes #2652
commit 35c2fa24 (2016-05-03): net/nut: Remove whitespace before conffiles

Closes #2652
closed 2016-05-03 by hnyman