openwrt/packages #2663: ldns: build error - try to install man pages

chris5560, 2016-05-06:
```
./libtool --tag=CC --quiet --mode=finish /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/lib
./install-sh  -d /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/bin
./install-sh  -c -m 755 packaging/ldns-config /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/bin
./install-sh  -d /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/man/man1
./install-sh  -c -m 444 ./packaging/ldns-config.1 /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/man/man1/
./install-sh  -d /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/man/man3
for f in doc/man/man3/*; do \
	./install-sh  -c -m 444 $f /home/owbuild/trunk/build_dir/target-i386_i486_musl-1.1.14/ldns-1.6.17/ipkg-install/usr/man/man3/; \
done
./install-sh: doc/man/man3/* does not exist.
Makefile:297: recipe for target 'install-manpages' failed
make[4]: *** [install-manpages] Error 1
```

k0ste, 2016-05-07:
@chris5560, hello. What distro in use?
This is not ldns trouble. It's an issue with Perl 5 ("Can't use 'defined(@array)'"), upstream should fix it.

On ArchLinux this avoid by this [patch](https://git.archlinux.org/svntogit/packages.git/tree/trunk/perl-recent.patch?h=packages/ldns&id=8ad1d1febe647fed60704bf68f880976adcb5ea7).

You can manually fix it:
build_dir/target/ldns-1.6.17/Makefile: on "install" & "uninstall" targets disble "install-manpages" & "uninstall-manpages".

chris5560, 2016-05-08:
Distro XUbuntu 16.04 x64 running inside VirtualBox
perl5 (revision 5 version 22 subversion 1

k0ste, 2016-05-09:
Yes, this version is affected. I make patch for avoid this, you should try.

chris5560, 2016-05-09:
Patch tested. Works fine.

commit a4a9038f (2016-07-03): ldns: Avoid perl bug for manpages.

Fix #2663

[refresh patch]
commit 47d41c53 (2016-08-12): ldns: Avoid perl bug for manpages.

Fix #2663

[refresh patch]

(cherry picked from commit a4a9038f58c4876d419a173bb17d34c0800fae75)
closed 2016-07-03 by champtar