openwrt/packages #2672: mc: can't build 4.8.17

k0ste, 2016-05-09:
After fix #2664:

```shell
Making all in po
make[6]: Entering directory '/home/k0ste/sandbox/openwrt/build_dir/target-mips_34kc_musl-1.1.14/mc-4.8.17/po'
*** error: gettext infrastructure mismatch: using a Makefile.in.in from gettext version 0.18 but the autoconf macros are from gettext version 0.19
Makefile:243: recipe for target 'check-macro-version' failed
```

Seems it from https://github.com/MidnightCommander/mc/commit/406765f0444e3d38ab18042cd5d7e47f977aff42

Fix by change **build_dir/target/mc-4.8.17/po/Makefile.in.in**:
```
GETTEXT_MACRO_VERSION
```

And build okay. Outside openwrt toolchain build is okay. Don't know right way how to fix this. May be some AM_GNU_GETTEXT_VERSION key in enivorment?

dibdot, 2016-05-09:
I've tested the build with different targets with normal openwrt toolchain (x86, ar71xx, bcm2709) without any problems ... also the buildbots have no problem with 4.8.17 - seems to be a problem in your environment.

k0ste, 2016-05-10:
@dibdot, thanks for feedback. I try to build for x86/bcm2709 - all okay. But for AR7xxx (TP-LINK TL-WR741N / 1043ND / whatever) - not.
I confused.

hnyman, 2016-05-13:
Possibly fixed by #2693

k0ste, 2016-05-15:
Works.

closed 2016-05-15 by k0ste