openwrt/packages #2687: ddns-scripts:Support for GoDaddy?

CarlEdman, 2016-05-12:
GoDaddy is one of the major US domain name registrars and offers a public API (https://developer.godaddy.com/getstarted) to update the records of domains one owns.  That makes it possible to use it as a ddns service, without going to any third-party solutions.  Unfortunately, their API does not exactly fit the pattern of the present ddns scripts.  To update the address record one must run the equivalent of:

`wget -q --method=PUT --body-data='[{"type":"A","name":"[NAME]","data":"[IP]","ttl":3600}]' --header 'Authorization: sso-key [USERNAME]:[PASSWORD]' --header 'Content-Type: application/json' https://api.godaddy.com/v1/domains/[DOMAIN]/records/A/home -O -`

Here, [NAME] is the host name and [DOMAIN] the domain name (i.e., to set address for 'home.foobar.xyz', [NAME]=home and [DOMAIN]=foobar.xyz).  [USERNAME] and [PASSWORD] are what is called the 'key' and 'secret' any GoDaddy customer can obtain at (https://developer.godaddy.com/keys/).  For it to work correctly, I had to use a production, not a test, key/secret.

The return upon success is an empty JSON list (i.e., the string "[]").

chris5560, 2016-05-12:
Only solution to support GoDaddy.com is to write a script similar to Cloudflair.com
Cloudflair is also using it's own API and JSON for update requests and answers.
I need someone who can do the testing for the script or write the script himself.
Then I do the integration into ddns-scripts (Makefile etc)
The conversation during development/testing can be done by direct mail.

CarlEdman, 2016-05-16:
Ok, I've written and tested such a script.  It is [here](https://github.com/CarlEdman/godaddy-ddns).  For further communication, feel free to email me directly using my username here at gmail.com.

yousong, 2016-08-11:
Guess ddns-scripts expects a shell script to avoid introducing an extra dependency on python which is big and slow on small devices.  Just open a pull request here when it's ready.

I am going to close this since it looks like doubts has be cleared and there is no update for almost 3 months

closed 2016-08-11 by yousong

CarlEdman, 2016-08-11:
The reason this is a python script is that @chris5560 had stated that the only solution is to write a script similar to that for Cloudflair.com  and the Cloudflair.com script is written in Python.  So spent some time cobbling together something as similar as possible to that.

But I agree entirely that something more lightweight, independent of python would be better.  That is why I originally suggested that one-line wget solution in opening this issue.

So I am happy to rewrite it to any requirement, as long as it is consistent.

yousong, 2016-08-11:
I guess Cloudflair.com is a typo and it should be cloudflare.com (https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/update_cloudflare_com.sh)?

It's a sad fact that we have so many open issues and pull requests and they just got ignored as time flows by without further words as we are such a loosely-coupled community.  I guess this is already the situation with the old trac ticket system of OpenWrt.

Anyway, feel free to reopen this or open a new one if you feel the need of it.

CarlEdman, 2016-08-11:
Thanks, @yousong for the pointer to the shell script.  I'll try to write something on that model in the next few days and will post a message here with a pointer once done.

As for reopening the issue, I do not see the interface for doing that.  Perhaps I do not have authority?

yousong, 2016-08-11:
Sorry, I thought original author can re-open his own issue, but it looks like github decides that this is not the case if the issue was closed by a repo collaborator.

However I think you should just open a new pull request against `master` of this repo, not a pointer, so that people can have proper review and discussions there.  Remember to AT the maintainer so that he can receive a notification of the event.

CarlEdman, 2016-08-11:
Thanks.  Will do.

michaudg, 2016-10-02:
If it can help, you can look at this bash script :  https://github.com/michaudg/godaddy-ddns

chris5560, 2016-10-02:
@michaudg
Thanks for your advise.

I already looked for it but I'm currently "fighting" with the json answers of godaddy.
I want to analyse their answers with json "tools" not with sed.
I'm using their test server api.ote-godaddy.com and jshn.sh from openwrt project.
I started with getting a domain list from /v1/domains but jshn.sh only answered:
"jshn: Failed to parse message data"
For me it looks that godaddy is missing some json parts inside their answer.
They answer with "[{ ... }{ ... }]".
From my knowledge an array inside json should be
{ objectname : [{ ... }{ ... }] }

Am I doing something wrong ?

yousong, 2016-10-08:
Object has to be the top type of json string for jshn to accept.   Maybe we can wrap the api response inside an object  `{'domains': <original-resp>}`, then parse it with `jshn`

jow-, 2016-10-08:
You could try using `jsonfilter`. Can you mention an example JSON fragment here  and tell me which values you like to extract?