openwrt/packages #2692: python File not found

linamu, 2016-05-12:
I run make package/python/compile V=99 and i get this 
what's wrong ?

mkdir -p /home/ubuntu/Bureau/openwrt/bin/ramips/packages /home/ubuntu/Bureau/openwrt/build_dir/target-mipsel_r2_uClibc-0.9.32/Python-2.7.11/ipkg-ramips/python-codecs/CONTROL
find /home/ubuntu/Bureau/openwrt/build_dir/target-mipsel_r2_uClibc-0.9.32/Python-2.7.11/ipkg-install -name "*\.pyc" -o -name "*\.pyo" | xargs rm -f
File not found '/home/ubuntu/Bureau/openwrt/build_dir/target-mipsel_r2_uClibc-0.9.32/Python-2.7.11/ipkg-install/usr/lib/python2.7/encodings+'
make[2]: *** [/home/ubuntu/Bureau/openwrt/bin/ramips/packages/python-codecs_2.7.11-8_ramips.ipk] Error 1

commodo, 2016-05-13:
did you do a full system build first ?

linamu, 2016-05-13:
yes i did ! a full system build before adding python package

linamu, 2016-05-13:
yes i did ! a full system build before adding python package

alose i have this message appear while compiling "cc1: note: someone does not honour COPTS correctly, passed 2 times" don't know if it causes the problem

commodo, 2016-05-13:
so, did you try building  Python 2.7.11 on a CC version of OpenWrt core ?
or are you building on top of trunk ?

commodo, 2016-05-13:
the "cc1: note: someone does not honour COPTS correctly, passed 2 times"   you can ignore for now ; they're just warnings

linamu, 2016-05-13:
i'm building on an old version of openwrt that i had its SDK from someone
i don't know if i understand well your question

commodo, 2016-05-13:
so, there are more versions of OpenWrt released;
Version CC (Chaos Calmer or 15.05), Backfire (I forgot the version number), Attitude Adjustment, White Russian, etc.

Python 2.7.11 was created to build on top of trunk, so an unreleased version of OpenWrt (version DD).
Python 2.7.9 works on top of version CC

So, officially  Python 2.7.11 was not tested on any version less newer than trunk.

linamu, 2016-05-13:
so i should just modify python version ? and try again ?

linamu, 2016-05-13:
so what should i change now ?

commodo, 2016-05-13:
well, you could try to make Python build for your environment;
i don't know what env you have to recommend anything

linamu, 2016-05-13:
in fact i'm using revision 25875 as i know

hnyman, 2016-05-13:
> in fact i'm using revision 25875 as i know

well, that is from year 2011 and completely unsupported. Using Linux kernel 2.6.3x etc. (while now kernel is 3.18 in CC15.05.1 and 4.1/4.4/+ in DD trunk)

Your router (and build system) need to be from CC15.05.1+ or DD trunk to have any relevance in the discussion here

yousong, 2016-08-02:
Close it for now as the version combination is not supported and the issue got no progress in the past 2 months.

closed 2016-08-02 by yousong