openwrt/packages #2702: Possible compilation problems with motion on ar71xx and x86

dfc-d, 2016-05-16:
@roger- 

It's kind of strange. Despite the following message in compile.txt (using build logs):

  Configure status
  motion 3.4.0-Git-78ba2da

OS : Linux
pthread support: No

** Fatal Error YOU MUST HAVE pthread Support *

The build completes successfully. I haven't yet had the opportunity to test if motion actually works and pthreads is gettinging linked despite configure not finding it, or if this is a case where 'bad things can happen at runtime'.

roger-, 2016-05-16:
Can you see if #2575 fixes this?

dfc-d, 2016-05-16:
Hi,

I actually created that patch, but closed it because it doesn't solve
the problem.  I haven't had the time to really dig into what the issue
is.  It is most peculiar.

It may only have building from SDK, I'm not sure.

Regards,

Daniel

roger-, 2016-05-17:
Ah sorry, missed that.

I haven't compiled motion in a while, so I'm assuming some recent change broke it. I'll try to do some investigation.

karlp, 2016-05-24:
related to https://github.com/openwrt/packages/issues/1814 ?

diizzyy, 2017-06-09:
@thess @hnyman 
Motion compiles, close.

closed 2017-06-09 by thess