openwrt/packages #2704: coreutils: installation does nothing

npenin, 2016-05-16:
Hi,

I wanted to have sed with unbuffered option so I decided to install coreutils, but it just does nothing.

I try to install coreutils on ar71xx, but the package just seems to do nothing at all. All the busybox bin are still here and nothing appears in /usr/bin. 

I think I have tried all the possible flags with `opkg`:
```opkg install coreutils --force-overwrite --force-reinstall --force-postinstall```

Could you please tell me what is wrong ?

Thanks,
Nicolas

thess, 2016-05-16:
There is no single "coreutils-all" package since there are over 60 separate stand-alone applets. The coreutils package is a dummy in order to handle the dependencies for building and packaging only. You will have to decide which ones you want and install them separately. From the package description:

>Full versions of standard GNU utilities. Normally, you would not use this
 package, since the functionality in BusyBox is more than sufficient and
 smaller.

closed 2016-05-16 by thess

npenin, 2016-05-16:
Can you then tell me where to find GNU sed then ? In the barrier breaker, I thought it was in this package.

thess, 2016-05-16:
Just the one in busybox is all we have. It looks like sed is still in 'oldpackages' but no one has cared enough to bring it forward from 14.07. Please open a new issue to request it or feel free to submit a PR and sign up as package maintainer.