openwrt/packages #2720: mwan3 incompatible with wifidog

pocha, 2016-05-20:
I recently setup mwan3 & Wifidog on TP Link 1043 ND with Openwrt version - CC. 

On user authentication Wifidog runs following commands 

    iptables -t mangle -A WiFiDog_br-lan_Incoming -d 192.168.9.157 -j ACCEPT
    iptables -t mangle -A WiFiDog_br-lan_Outgoing -s 192.168.9.157 -m mac --mac-source a0:99:9b:02:8b:19 -j MARK --set-mark 2

Now the commands are suppose to whitelist the client `192.168.9.157` so the internet starts to work. Ping from the client still keep getting dropped & I suspect its something to do with mwan. 

[mwan3 wiki for nodogsplash](https://wiki.openwrt.org/doc/howto/mwan3#nodogsplash) talks about changing the marking scheme in nodogsplash to make it compatible with mwan3. While wifidog does not expose these values in its config file, I did try changing the values in the code header file but it did not help. 

Looking at mwan3 code, it looked like mwan3 only picks up higher 8 bits of the packet mark as at most places the code says something like 

    -m mark --mark 0x0/0xff00

Wifidog is putting a mark `0x2` for known users which should not get caught by mwan3 logic. Also, looks like mwan3 is dropping the packets thinking them to be for a wan interface that actually does not exist. 

I am pretty sure this is game of marks itself. If someone has previously dealt with mwan3 or wifidog, I could use some help.

asterix555, 2017-02-01:
Hi Pocha,

thank you for your post. Tried it to solve it the same way - without success.
I setup a separate device with wifidog and the real router with mwan3 as a workaround.  

Let us know if you (or anybody else) made some progress on this issue!

regards

pocha, 2017-02-01:
I had to change the wifidog code where it looks for packet mark 0x2 & replaced it with 0x2/0x00ff. It solved the problem for me as wifidog is now looking for mark 0x2 in lower 8 bits only while mwan3 marks in higher 8 bits.

asterix555, 2017-02-01:
Thanks for your fast response!
Can you describe your patch a little closer? I don't find the corresponding code :-(

`wifidog-1.2.1$ find . -type f -exec grep -H FW_MARK_KNOWN {} \;
./patches/010-mangle-marks:-    FW_MARK_KNOWN = 2,  /**< @brief The client is known to the firewall */
./patches/010-mangle-marks:+    FW_MARK_KNOWN = 262144,  /**< @brief The client is known to the firewall */
./src/auth.c:        fw_allow(client, FW_MARK_KNOWN);
./src/firewall.h:    FW_MARK_KNOWN = 2,  /**< @brief The client is known to the firewall */
./src/firewall.c:                    if (tmp->fw_connection_state != FW_MARK_KNOWN) {
./src/firewall.c:                        fw_allow(tmp, FW_MARK_KNOWN);
./src/fw_iptables.c:                            FW_MARK_KNOWN);
./src/fw_iptables.c:                            " -p tcp --dport 80 -m mark --mark 0x%u -j REDIRECT --to-port %u", FW_MARK_KNOWN,
./src/fw_iptables.c:    iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j ACCEPT", FW_MARK_KNOWN);
./src/fw_iptables.c:    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j " CHAIN_KNOWN, FW_MARK_KNOWN);`

pocha, 2017-02-01:
-m mark --mark 0x℅u/0x00ff

asterix555, 2017-02-01:
thank you,
I applied this patch - and it works fine now!

thanks again!!
```
Index: wifidog-1.2.1/src/commandline.c
===================================================================
--- wifidog-1.2.1.orig/src/commandline.c
+++ wifidog-1.2.1/src/commandline.c
@@ -73,6 +73,8 @@ usage(void)
     fprintf(stdout, "  -i <path>     Internal socket path used when re-starting self\n");
     fprintf(stdout, "  -a <path>     Path to /proc/net/arp replacement - mainly useful for debugging.\n");
     fprintf(stdout, "\n");
+    fprintf(stdout, "Patched version for mwan3-compatibility\n");
+    fprintf(stdout, "\n");
 }

 /** Uses getopt() to parse the command line and set configuration values
Index: wifidog-1.2.1/src/fw_iptables.c
===================================================================
--- wifidog-1.2.1.orig/src/fw_iptables.c
+++ wifidog-1.2.1/src/fw_iptables.c
@@ -317,22 +317,22 @@ iptables_fw_init(void)
     if ((proxy_port = config_get_config()->proxy_port) != 0) {
         debug(LOG_DEBUG, "Proxy port set, setting proxy rule");
         iptables_do_command("-t nat -A " CHAIN_TO_INTERNET
-                            " -p tcp --dport 80 -m mark --mark 0x%u -j REDIRECT --to-port %u", FW_MARK_KNOWN,
+                            " -p tcp --dport 80 -m mark --mark 0x%u/0x00ff -j REDIRECT --to-port %u", FW_MARK_KNOWN,
                             proxy_port);
         iptables_do_command("-t nat -A " CHAIN_TO_INTERNET
-                            " -p tcp --dport 80 -m mark --mark 0x%u -j REDIRECT --to-port %u", FW_MARK_PROBATION,
+                            " -p tcp --dport 80 -m mark --mark 0x%u/0x00ff -j REDIRECT --to-port %u", FW_MARK_PROBATION,
                             proxy_port);
     }

-    iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j ACCEPT", FW_MARK_KNOWN);
-    iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j ACCEPT", FW_MARK_PROBATION);
+    iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j ACCEPT", FW_MARK_KNOWN);
+    iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j ACCEPT", FW_MARK_PROBATION);
     iptables_do_command("-t nat -A " CHAIN_TO_INTERNET " -j " CHAIN_UNKNOWN);

     iptables_do_command("-t nat -A " CHAIN_UNKNOWN " -j " CHAIN_AUTHSERVERS);
     iptables_do_command("-t nat -A " CHAIN_UNKNOWN " -j " CHAIN_GLOBAL);
     if (got_authdown_ruleset) {
         iptables_do_command("-t nat -A " CHAIN_UNKNOWN " -j " CHAIN_AUTH_IS_DOWN);
-        iptables_do_command("-t nat -A " CHAIN_AUTH_IS_DOWN " -m mark --mark 0x%u -j ACCEPT", FW_MARK_AUTH_IS_DOWN);
+        iptables_do_command("-t nat -A " CHAIN_AUTH_IS_DOWN " -m mark --mark 0x%u/0x00ff -j ACCEPT", FW_MARK_AUTH_IS_DOWN);
     }
     iptables_do_command("-t nat -A " CHAIN_UNKNOWN " -p tcp --dport 80 -j REDIRECT --to-ports %d", gw_port);

@@ -374,21 +374,21 @@ iptables_fw_init(void)
     iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -j " CHAIN_AUTHSERVERS);
     iptables_fw_set_authservers();

-    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j " CHAIN_LOCKED, FW_MARK_LOCKED);
+    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j " CHAIN_LOCKED, FW_MARK_LOCKED);
     iptables_load_ruleset("filter", FWRULESET_LOCKED_USERS, CHAIN_LOCKED);

     iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -j " CHAIN_GLOBAL);
     iptables_load_ruleset("filter", FWRULESET_GLOBAL, CHAIN_GLOBAL);
     iptables_load_ruleset("nat", FWRULESET_GLOBAL, CHAIN_GLOBAL);

-    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j " CHAIN_VALIDATE, FW_MARK_PROBATION);
+    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j " CHAIN_VALIDATE, FW_MARK_PROBATION);
     iptables_load_ruleset("filter", FWRULESET_VALIDATING_USERS, CHAIN_VALIDATE);

-    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j " CHAIN_KNOWN, FW_MARK_KNOWN);
+    iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j " CHAIN_KNOWN, FW_MARK_KNOWN);
     iptables_load_ruleset("filter", FWRULESET_KNOWN_USERS, CHAIN_KNOWN);

     if (got_authdown_ruleset) {
-        iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u -j " CHAIN_AUTH_IS_DOWN,
+        iptables_do_command("-t filter -A " CHAIN_TO_INTERNET " -m mark --mark 0x%u/0x00ff -j " CHAIN_AUTH_IS_DOWN,
                             FW_MARK_AUTH_IS_DOWN);
         iptables_load_ruleset("filter", FWRULESET_AUTH_IS_DOWN, CHAIN_AUTH_IS_DOWN);
     }

```

diizzyy, 2017-06-09:
@pocha @asterix555 
Fixed? Can we close this?

pocha, 2017-06-09:
Ya - this is an issue with wifidog not mwan3

closed 2017-06-09 by pocha

zimiao815, 2018-10-18:
yes，  thanks ! i got it too