openwrt/packages #2725: privoxy conffiles

zyxmon, 2016-05-22:
https://github.com/openwrt/packages/blob/master/net/privoxy/Makefile#L114
should be `/etc/privoxy/config`

hnyman, 2016-05-22:
> should ``` be /etc/privoxy/config ```

Why do you think so?
On line 155 Makefile installs the config files as ``` /etc/config/privoxy ```
https://github.com/openwrt/packages/blob/master/net/privoxy/Makefile#L155

hnyman, 2016-05-22:
@chris5560

zyxmon, 2016-05-22:
> The
       default configuration files are: config, default.filter, default.action
       and default.action. user.action should  be  used  for  locally  defined
       exceptions to the default rules in match-all.action and default.action,
       and user.filter for locally defined filters. These are well  commented.
       On  Unix  and  Unix-like systems, these are located in /etc/privoxy/ by
       default.

chris5560, 2016-05-22:
You are looking into Makefile of Privoxy source not of the OpenWrt package.

The default configuration files are still located at their default location /etc/privoxy/ except config.

The reason for this is, that OpenWrt is using UCI. So the configs are stored in /etc/config/ for all packages using the same (UCI) syntax. This config files are also used by LuCI web user interface.
So /etc/privoxy/config does not exist or is ignored if exist.
On every startup /etc/init.d/privoxy creates /var/etc/privoxy.conf from /etc/config/privoxy and handover to privoxy as configuration file to use.

hnyman, 2016-05-22:
Thanks for explanation. Not a bug.

closed 2016-05-22 by hnyman

zyxmon, 2016-05-22:
My mistake, sorry. There was a bug with prvoxy config in Entware that I extended to openwrt.