openwrt/packages #2748: smbus python3 - no option in menu, but package can be compiled

nekitip, 2016-05-25:
There is a smbus python package in **i2c-tools.**
The same file (from openwrt) can also be compiled to work with python3 in regular debian x64 by simply "python3 setup.py build" on that same file. However it's near to imposible to do it this way in openwrt toolchain (crosscompile) by hand.
Openwrt makemenu already has option to build package and it builds fine and works, but it builds in python 2.7. So, the only thing needed to change are options for **makemenu**, so python 3.x can utilise this package.
Unfortunately, I have failed to make those changes myself... :(

commodo, 2016-06-04:
so, i2c-tools is packaged to work with `python`
to work with python3,it should be packaged to build with python3 bindings; 

it does not look too hard to package it;
but we'd need to get the maintainer to look into it, or you can also update the i2c-tools package and submit a PR
@dangowrt  ?

nekitip, 2016-06-05:
Like I said - I'm not that experienced. Yes, I can post here the makefile that does compile i2c smbus package into python3 module, but after build, something fails (in my opinion, maybe install dir?), so it has to manually be copied to the right location on target device.
As for smbus c file itself, I'm not sure if mine is not working because I've done somethign wrong in makefile or it is not working at all, because **when you open it** (it's not that big file, try and look at the end) - you will see that someone made a lot of IF python3 cases, wich led me to belive that this .c file can indeed be compiled to python3 without messing with it, but I failed. So, I found on various forums different .c file, sort of a "patched file", that I managed to compile with my changed makefile.
So, if you lost me along the way - here is what I suggest:
-let someone who knows what to do, to first try to create standard openwrt option in makefile envirement for python3 i2c smbus driver, and see if it smbus compiles.
If not, then I will give both my makefile (and you fix it all the way), and this patched .c file I found on various forums.

dangowrt, 2016-11-17:
Should be fixed by https://github.com/openwrt/packages/pull/3073
If so, please close this ticket.

diizzyy, 2017-06-09:
@dangowrt 
Can you close this due to reporter timeout?

closed 2017-06-20 by nekitip